microsoft/Azure-Kinect-Sensor-SDK #867: about Multiple Device Sync Issues

nonlinear1, 2019-10-31:
I have run two kinects (K4A) in one program with external synchronized units. But the two kinects just can run for a while and then the program crashed. At the sametime, the terminal give me the following information:

[2019-10-31 08:56:02.662] [error] [t=12454] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/allocator/allocator.c (118): k4a_capture_t_get_context(). Invalid k4a_capture_t (nil)
[2019-10-31 08:56:02.662] [error] [t=12454] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/allocator/allocator.c (295): Invalid argument to capture_dec_ref(). capture_handle ((nil)) is not a valid handle of type k4a_capture_t
[2019-10-31 08:56:02.676] [error] [t=12454] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (158): depthmcu_get_serialnum(). depthmcu_get_serialnum found non-printable serial number (character 0 is ASCII value 1)
[2019-10-31 08:56:02.676] [error] [t=12454] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth/depth.c (349): depthmcu_get_serialnum(depth->depthmcu, serial_number, serial_number_size) returned failure in depth_get_device_serialnum()
[2019-10-31 08:56:02.676] [error] [t=12454] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (958): depth_get_device_serialnum(device->depth, serial_number, serial_number_size) returned failure in k4a_device_get_serialnum()
Failed to get serial number length
Segmentation fault (core dumped)

the environment for my program:
hardware:
cpu i7 
ram 16G
GPU nvidia gtx 1070

the software enviroment:
os: ubuntu18.04
sdk1.3.0
Device Serial Number: 000170693412
Current Firmware Versions:
  RGB camera firmware:      1.6.102
  Depth camera firmware:    1.6.75
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft
I have set usbfs_memory_mb=1000 according to https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/485
at the same time if I run the two kinect in k4aviewer in syn mode, the two k4aviewer don't crash.

Could someone know why, and what does mean the above error message means?
Thank you very much!

wes-b, 2019-10-31:
A guess, you need to set synchronized_images_only to true, or NULL check your images to ensure you received valid images.

nonlinear1, 2019-10-31:
Thank you for your help. In fact, I have set synchronized_images_only = true.  the question is my program carshed and  has the following information:
"[2019-10-31 08:56:02.676] [error] [t=12454] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (958): depth_get_device_serialnum(device->depth, serial_number, serial_number_size) returned failure in k4a_device_get_serialnum()
Failed to get serial number length
Segmentation fault (core dumped)"
It don't give me any null image and just carshed, I have nothing to do, for the error message comes from k4a sdk.

wes-b, 2019-10-31:
Your app crashed so you are going to need to get it on the debugger. In your first post there is an error about a NULL capture_t, so you should try to figure out where that is coming from. Definitely focus on the first error your are seeing, not the one before the crash.

nonlinear1, 2019-10-31:
Thank you very much

nonlinear1, 2019-10-31:
＠wes-b
Just now, I put my program into the examples folder in Azure-Kinect-Sensor-SDK. Then I build the Azure-Kinect-Sensor-SDK with debug build "cmake .. -GNinja -DCMAKE_BUILD_TYPE=Debug". Then I use gdb to run my program. My program run for a while, and then crashed. the terminal give me the following information:
[device.c:914/uvc_release_if] reattached kernel driver to interface 0
[device.c:923] end uvc_release_if (0)
[device.c:1746/_uvc_status_callback] libusb_submit_transfer() = -1
[device.c:1748] end _uvc_status_callback
[device.c:837] begin uvc_unref_device
[Thread 0x7fffdbfff700 (LWP 31071) exited]
[device.c:845] end uvc_unref_device
[device.c:1516] begin uvc_free_devh
[device.c:419] begin uvc_free_device_info
[device.c:459] end uvc_free_device_info
[device.c:1526] end uvc_free_devh
[device.c:1564] end uvc_close
[device.c:837] begin uvc_unref_device
[device.c:845] end uvc_unref_device
[Thread 0x7fffb4ec0700 (LWP 31087) exited]
[Thread 0x7fffb5ffc700 (LWP 31086) exited]
[Thread 0x7fffb67fd700 (LWP 31085) exited]
[Thread 0x7fff8dffb700 (LWP 31096) exited]
[Thread 0x7fff8e7fc700 (LWP 31095) exited]
[Thread 0x7fff8effd700 (LWP 31094) exited]
[Thread 0x7fff8f7fe700 (LWP 31093) exited]
[Thread 0x7fff8ffff700 (LWP 31092) exited]
[Thread 0x7fffa4ffb700 (LWP 31091) exited]
[Thread 0x7fffa57fc700 (LWP 31090) exited]
[Thread 0x7fffc60b4700 (LWP 31082) exited]
[Thread 0x7fffc6ffd700 (LWP 31081) exited]
[Thread 0x7fffc77fe700 (LWP 31080) exited]
[Thread 0x7fffc7fff700 (LWP 31079) exited]
[Thread 0x7fffd1486700 (LWP 31078) exited]
[Thread 0x7fffd1c87700 (LWP 31077) exited]
[Thread 0x7fffda8f4700 (LWP 31076) exited]
[Thread 0x7fffdb0f5700 (LWP 31075) exited]
[Thread 0x7fffe0dc4700 (LWP 31064) exited]
[Thread 0x7fffe17ae700 (LWP 31061) exited]
[Thread 0x7ffff7f9c580 (LWP 31056) exited]
[Inferior 1 (process 31056) exited with code 0377]
(gdb) bt
No stack.
(gdb) 
No stack, I cann't find any information about the reason of crash. 
Could you have any suggestion? Thanks a lot.

nonlinear1, 2019-10-31:
I have solve this problem! Thank you webs-b

closed 2019-10-31 by wes-b

wirthual, 2019-11-01:
Hi,
can you tell what was the issue and how you solved it? I ran into the same problems.

nonlinear1, 2019-11-03:
@wirthual 
First, you shoud update your firmware to the latest one!
second, you should check "k4a_capture_release", this release the object is right?
for me the k4a_capture_release release the object , which is no capture.

DreamOneYou, 2021-02-03:
Hi, I also encountered the same problem, but I don’t understand your solution very well. Can you tell me more about it? This is my mistake.
[2021-02-03 16:01:40.184] [error] [t=17228] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\allocator\allocator.c (118): k4a_capture_t_get_context(). Invalid k4a_capture_t 000002698D1926B0
[2021-02-03 16:01:40.184] [error] [t=17228] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\allocator\allocator.c (344): Invalid argument to capture_get_color_image(). capture_handle (000002698D1926B0) is not a valid handle of type k4a_capture_t