microsoft/Azure-Kinect-Sensor-SDK #869: User access to Kinect

ETowleMS, 2019-10-31:
(merging from User Voice request:  https://feedback.azure.com/forums/920053/suggestions/38915212) 

Is there a way to enable user access to the tools (k4aviewer) from a user account without sudo? I tried creating a rules file in /etc/udev/rules.d giving 666 access to the users for all 5 usb devices the kinect uses but I still get the error below. (works fine if I sudo)

[2019-10-29 11:13:09.740] [critical] [t=7522] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (360): find_libusb_device(). libusb device(s) are all unavalable. Is the device being used by another application?
[2019-10-29 11:13:09.741] [error] [t=7522] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2019-10-29 11:13:09.741] [error] [t=7522] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()

Brent-A, 2019-10-31:
If you create the rules file from [these instructions](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md) it should work without sudo. If you still have trouble we may need more details on what you are changing in rules.d specifically.

closed 2019-10-31 by Brent-A

petrasovaa, 2020-02-11:
Hi, I have the exact same problem, I used the rules file from this repo and rebooted and reconnected. With sudo everything works. Testing this on new Ubuntu 18.04. Anything else to try?

wes-b, 2020-02-11:
So far this has worked for everyone, so double check your steps.

Also review other rules files that might also be running. It looks like some users may also have rules.d in /usr/lib/udev/rules.d that get merged in.

petrasovaa, 2020-02-12:
You were right, totally my mistake, sorry for the noise.