microsoft/Azure-Kinect-Sensor-SDK #871: ARM support for Azure Kinect

tesych, 2019-10-31:
**Is your feature request related to a problem? Please describe.**
Both Sensor and Body Tracking SDK cannot do not have support for ARM.

**Describe the solution you'd like**
Support the Azure Kinect Sensor SDK for Linux on ARM on NVIDIA Jetson Nano or similar embedded dev platforms.

**Additional context**
This feature is in planning stage for the implementation and we want our community feedback! The [PR with the proposal for ARM support](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/pull/870) is created and ready for review.

nonlinear1, 2019-11-02:
Please , give support for Nvidia Jetson Xavier on Ubuntu 18.04, I have been waitting for several months! For slam and other application such as AutoPilot，the jetson nano and jetson tx2 is lower computing ability than Xavier. It cann't run some slam and AutoPilot apps and at the same time, it limit the kinect for azure in slam and AutoPilot. Just my idea.
Thanks a lot

tesych, 2019-11-18:
@nonlinear1, thank you for the ask. We will definitely consider Nvidia Jetson Xavier as well.

tkircher, 2020-01-09:
Is there a significant issue with just recompiling the depth library for ARM? From the disassembly it looks like DepthTransforms::DepthEngineCalibration, DepthTransforms::CpuImageProcessing, etc. are pretty straightforward. If the OpenGLDepthEngine compiles using standard OpenGL libraries on Linux there should be no trouble. At least start out with an empty library with stubs for the function pointers in k4a_plugin.

compiaffe, 2020-02-26:
@tesych Would you be able to share a status update on ARM support for the sensor SDK? Thank you.

bhazza, 2020-02-27:
+1 I'd be interested in know this too. Thanks!

airfield20, 2020-02-28:
Also Following

nonlinear1, 2020-03-18:
Could you tell me when you will publish the arm sdk of kinect for azure?

wes-b, 2020-03-18:
Tracked with #1093. Beta is out now.

jlnacyp, 2020-04-28:
The Body Tracking SDK does not support ARM？About how long can support.

uersoy, 2020-05-21:
> @nonlinear1, thank you for the ask. We will definitely consider Nvidia Jetson Xavier as well.

Nvidia Jetson Xavier NX has been released to public and many people will be switch to it soon. Are you still considering it?

jpj625, 2020-07-23:
@wes-b Any update on ARM support for body tracking SDK? I can't find any open issues, and #822 which might be related/blocking is 9 months stale.

My journey of discovery, for anyone else looking:
- The [Support for ARM proposal](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/proposals/ARM-support.md) appears to include both Sensor and BT, but it's been 4 months since the 1.4.0 release
- [This commit](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/commit/cc9e91f29e6a1258f7f1a61ec660a9a44db885d8) in 1.4.1 marks the proposal as feature complete
- #1093/[a comment](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1093#issuecomment-611617173) indicates that BT compatibility with 1.4.x is on a separate track
- #1093/[a comment](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1093#issuecomment-597955591) from 4 months ago says that #822 has "technical issues" and "will release as soon as possible"

qm13, 2020-07-27:
Sensor SDK has ARM support. We are in the process of adding ARM support to Body Tracking SDK.

marcfischer123467, 2020-07-29:
Will there be support for HoloLens 2 ?

Fred3D-tech, 2020-08-15:
is there a documentation to install the azure kinect on a jetson xavier agx ?

tkircher, 2020-08-16:
> is there a documentation to install the azure kinect on a jetson xavier agx ?

If you're running Jetpack 4.4, it's the same as installing on any other Jetson.

weidongshao, 2020-12-30:
Ping on the status of body tracking on ARM (Jetson Xavier). Is there an estimate release date?

ShravanRai22, 2021-01-05:
Is the body tracking sdk supported in Nvidia Jetson AGX Xavier?

thinko, 2021-04-08:
It's been another few months, any update on this? Xavier NX and AGX Xavier bring a lot more horsepower to the table -- it seems like that was something holding this back previously.

weidongshao, 2021-04-09:
> It's been another few months, any update on this? Xavier NX and AGX Xavier bring a lot more horsepower to the table -- it seems like that was something holding this back previously.

It seems this is not a priority for MSFT, there is no incentive to empower a competing hardware platform. What is your use case for body tracking SDK? I would suggest that you try OpenPose or other AI models using Nvidia's toolkit

carloshgaray, 2021-04-21:
Ping on the status of body tracking on ARM (Jetson Xavier). Is there an estimated release date?

leodesrumaux, 2021-04-24:
Same question : Ping on the status of body tracking on ARM (Jetson Xavier). Is there an estimated release date? @tesych

qm13, 2021-04-26:
There is no committed date. The work is ongoing.

nlemoult, 2021-04-30:
I join the discussion to also show my interest for body tracking support on a Jetson Xavier NX / ARM64 platform. 

We are developing an embedded solution that would really need this support. Happy to learn that the work is underway. Despite the lack of a beta release date, this is already interesting news.

Promit, 2022-11-01:
I'd like to bump this again, with the release of the Windows Dev Kit 2023, and what appears to be a much more powerful hardware platform.

tkircher, 2022-11-01:
> I'd like to bump this again, with the release of the Windows Dev Kit 2023, and what appears to be a much more powerful hardware platform.

There also hasn't been a support update in two years, despite several known issues that could be fixed, and performance improvements that have been identified.

bhomaidan1990, 2023-04-01:
We are in Apr 2022, and I can't find the support for Arm64, can you please tell me if there is a way/workaround to install BT on Xavier NX-Dev kit?? thanks.

tank104, 2023-04-01:
2023 even …. :)

tkircher, 2023-04-02:
Rumor has it that Microsoft is focused on Hololens 3, Mesh, and MRTK and has soft-canceled the Kinect. So don't expect any more releases or additional support or functionality. Might be a good time for someone to add BT for Kinect to OpenCV.

PR #1108 (2020-03-06): Arm64 Support
## Fixes #871

### Description of the changes:
- Adding ARM64 Support