microsoft/Azure-Kinect-Sensor-SDK #873: Depth measure problem and a weird point cloud

lihk11, 2019-11-01:
I am not sure it is a bug or not, but it really confused me.
I am using the Azure Kinect DK to get point cloud for some parts, but the point clouds I get are weird, please take a look at the following picutre, 
[image]

 The middle one is correct, but the parts in both sides seem to have a dirtortion, because the middle stick of the part is not at the middle...
 
the middle part(expected result):
[image]

the left one (the middle stick is more left than expected):
[image]
the left one at a different view angle:
[image]

Could someone point out what the problem is?

rabbitdaxi, 2019-11-04:
@lihk11 Thank you for the feedback. It is a little bit challenging to see what is the exactly issue by looking at this point cloud rendering without looking at the real object/environment or setup... Would it be possible for you to make a mkv recording by using the k4arecorder from the SDK when you can repro this issue and share to us? If you feel not comfortable share the recording, then we can  using this issue channel to discuss more by guessing what might be causing it.

jesd23, 2019-11-04:
Not and check zillow. _vneo,a k popsicleface, james e smith also confused by communication to bring new color.

rabbitdaxi, 2019-12-02:
@lihk11 please let us know whether you have information that can help clarify the issue here. See previous post. For issues more than 7 days old without response, we will close them.

closed 2019-12-03 by rabbitdaxi

jesd23, 2020-02-01:
********PRIVATE Confidential Information *******.