microsoft/Azure-Kinect-Sensor-SDK #878: Failed to start device...

magichome, 2019-11-02:
When running k4aviewer.exe, the Kinect device would open but fail to start.  I resolved the issue but wanted to capture my findings here so it may help others.

System:
```
Lenovo 80VF
Windows 10 Pro v10.0.18362
```
Log output:
```
[2019-11-02 14:49:04.614] [error] [t=2108] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\color\mfcamerareader.cpp (410): Start(). Failed to set media type at start: 0x800706bf
[2019-11-02 14:49:04.614] [error] [t=2108] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\color\color.cpp (194): color->m_spCameraReader->Start(width, height, fps, config->color_format, &color_capture_available, color) returned failure in color_start()
[2019-11-02 14:49:04.614] [error] [t=2108] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (905): color_start(device->color, config) returned failure in k4a_device_start_cameras()
```
Additionally, the Event Viewer -> Windows Logs -> System reported:
```
The Windows Camera Frame Server service terminated unexpectedly.
```

Solution:
```
1. Create a Restore Point first.
2. Start Regedit.exe and go to the following key:

HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows Media Foundation\Platform

3. Create a REG_DWORD value named EnableFrameServerMode and leave the value data for EnableFrameServerMode to 0.
4. If you’re using Windows 64-bit, create the REG_DWORD value named EnableFrameServerMode also in the following location:
 
HKEY_LOCAL_MACHINE\SOFTWARE\WOW6432Node\Microsoft\Windows Media Foundation\Platform

The above key is for 32-bit applications running on a Windows 64-bit computer.
5. Exit the Registry Editor.
6. Reboot
```
This solution was offered by 'paolone' on the Lenovo support forum here: https://forums.lenovo.com/t5/ThinkPad-T400-T500-and-newer-T/KB4284835-causes-webcam-to-stop-on-T470p/td-p/4110977/page/2

closed 2019-11-02 by magichome

wes-b, 2019-11-04:
Good to know, thanks for sharing. 

Usually when users see:
>Start(). Failed to set media type at start: 0x800706bf
They can open up the 'Event Viewer', where they will find that frame server is crashing and the binary name that is causing the crash.

#853, #381

amusedamused, 2022-12-06:
Hi , 
I'm having problem with the same issuei think , ive been trough the whole trouble shooting, made sure  i have usb 3.0, tried all usb ports, rebooting and power cykling, the kinect shows up in both devices and the kinect wiever, lights as they should be, but cant start depth camera, color camera working just fine. Everything was working with body tracking as well just a couple a days ago, i tried it with the Azure Kinect Examples for Unity, left the kinect plugged into the computer for some days ( could this have caused some burn of the lazer? unity was not running the build for this time. ) I'm quite desperate as i was unsing it for a work demo for a futur job. This is the error i recive in the kinect wiever. I have not reset it to facory reset as it was working with the current firmware just some days ago but maybe I should try? 

[ error    ] : usb_cmd_io(). Command(00000022) sequence ended in failure, TransactionId 55555555 == 0000000A Response size 0x00000010 == 0x00000010 Packet status 0x55555555 == 0x00000000 Packet type 0x55555555 == 0x0a6fe000
[ error    ] : usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, p_data, data_size, NULL, 0, bytes_read, &cmd_status) returned failure in usb_cmd_read()
[ error    ] : usb_cmd_read( depthmcu->usb_cmd, DEV_CMD_NV_DATA_GET, (uint8_t *)&nv_tag, sizeof(nv_tag), calibration, cal_size, bytes_read) returned failure in depthmcu_get_cal()
[ error    ] : depthmcu_get_cal(depth->depthmcu, depth->calibration_memory, DEPTH_CALIBRATION_DATA_SIZE, &depth->calibration_memory_size) returned failure in depth_start()
[ error    ] : usb_cmd_io(). Command(000000F2) sequence ended in failure, TransactionId 55545555 == 0000000B Response size 0x00000010 == 0x00000010 Packet status 0x55555555 == 0x00000000 Packet type 0x55555555 == 0x0a6fe000
[ error    ] : usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, NULL, 0, p_data, data_size, NULL, cmd_status) returned failure in usb_cmd_write_with_status()
[ error    ] : usb_cmd_write_with_status(depthmcu->usb_cmd, DEV_CMD_DEPTH_STREAM_STOP, NULL, 0, NULL, 0, &cmd_status) returned failure in depthmcu_depth_stop_streaming()
[ error    ] : usb_cmd_io(). Command(0000000A) sequence ended in failure, TransactionId 55555455 == 0000000C Response size 0x00000010 == 0x00000010 Packet status 0x55555555 == 0x00000000 Packet type 0x55555554 == 0x0a6fe000
[ error    ] : usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, NULL, 0, p_data, data_size, NULL, cmd_status) returned failure in usb_cmd_write_with_status()
[ error    ] : usb_cmd_write_with_status(depthmcu->usb_cmd, DEV_CMD_DEPTH_STOP, NULL, 0, NULL, 0, &cmd_status) returned failure in depthmcu_depth_stop_streaming()
[ error    ] : depth_start(device->depth, config) returned failure in k4a_device_start_cameras()

amusedamused, 2022-12-06:
[image]
this is the firmware im having installed.