microsoft/Azure-Kinect-Sensor-SDK #879: Issue with installation

rajk97, 2019-11-02:
Hi all! I'm trying to install k4atools. When I run

[sudo apt install k4a-tools](url)

I get the following error:

[Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 k4a-tools : Depends: libsoundio1 (>= 1.0.1) but it is not installable
E: Unable to correct problems, you have held broken packages.
](url)

Please help me out. I'm a beginner to this. I'm using Ubuntu 18.04. Thanks.

Brent-A, 2019-11-04:
(Edited original post for formatting)

schultetwin1, 2019-11-04:
`k4a-tools` includes a binary called `k4a_viewer` that takes a dependency on libsoundio. The `(>= 1.0.1)` part means we need a version of libsoundio that is greater than 1.0.1.

There are many tutorials out there for installing packages after hitting this error. The first suggestion on all of them is using `-f` in the apt-get install command. I'd search around and see if you can find a way around this.

rajk97, 2019-11-04:
Thank you for your response. I'll look into it.

*Raj *Kolamuri
Graduate Student
Mechanical Engg.
Carnegie Mellon University
Ph: +412 897 8877
EMail: rkolamur@andrew.cmu.edu

ᐧ

closed 2019-11-13 by wes-b