microsoft/Azure-Kinect-Sensor-SDK #882: Choose the one to run body tracking if there are more than one GPU devices

wqs111000, 2019-11-04:
Body tracking needs a large amount of GPU computation. The computer may have more than one GPU devices. When one GPU is busy,  Body tracking had better switch to use another GPU.

So an feature that  the tracker can choose the  GPU device to run body tracking is requested. 
If the feature or something like it is already available, please show me.Thanks.

**Describe the solution you'd like**
Maybe add a new variable to k4abt_tracker_configuration_t like cpu_only_mode.

yijiew, 2019-11-05:
@wqs111000 : Thanks for the feedback. This feature is not available. Just out of curiosity, what other apps are you running when you are running this body tracking SDK. Why can't you switch the other app to run on the other GPU instead of switching the body tracking app?

wqs111000, 2019-11-06:
Thanks for your reply. Yes, to switch the other app is also a choice. But if the body tracking SDK can also do it, the allocation of  GPUs  sometimes will be much easy. For example, I have two  GPU devices with quite different performance and the computer has some flexible assignments ( may do or not  do,not sure ). Under current conditions, body tracking had to run on the bad GPU even when the good GPU is unoccupied at some cases.

yijiew, 2019-11-06:
@wqs111000 I see. Thanks for the detailed explanation. I will discuss among our team about adding this feature for the next release. Please stay tuned :)

diablodale, 2019-11-09:
To share my own answer to @yijiew question 'why can't you switch the other app'...it is because apps often ask the same of other apps. Leading to no apps allowing choice of GPU. Classic finger-pointing circular scenario.
It could also be the need to chose a specific shared GPU so that two different processes have access to the same OpenGL resources (e.g. shared OpenGL contexts)
In this century's choice of power management and multiple GPUs, it is common for a computing device to have need to manage app usage of GPU, CPU, and power resources. Optimizing and **testing** them for specific repeatable behavior. Dynamic/random assignment of a GPU is not reliable and instead requires specific choice for solid tested solutions.

wqs111000, 2019-11-18:
@yijiew : Hi! What is default principle according to which the body tracker chooses GPU device.Does it always use the best-performing one among several different GPU device?

yijiew, 2019-11-18:
@wqs111000 Unfortunately it is not. The default principle is to use GPU 0. 

We had discussed this issue internally and decided to add a config setting that allows the user to choose the GPU device to use. The feature will be added in the next release. Please stay tuned.

wqs111000, 2019-12-06:
@yijiew :Thanks a lot. I find the feature have been available in Body Tracking SDK v0.9.5.

yijiew, 2019-12-06:
@wqs111000 : You are welcome. Please let us know if you hit other issues when turning on this feature.

rfilkov, 2019-12-06:
@wqs111000 @yijiew Is v0.9.5 available for download? I don't see it here yet: https://docs.microsoft.com/en-us/azure/Kinect-dk/body-sdk-download

yijiew, 2019-12-06:
Yes! It is available already. But for some reason, the doc page is not updated yet. You can download the nuget package from: https://www.nuget.org/packages/Microsoft.Azure.Kinect.BodyTracking/
msi from: https://www.microsoft.com/en-us/download/details.aspx?id=100636.

qm13, 2019-12-07:
Fixed in v0.9.5.

closed 2019-12-07 by qm13