microsoft/Azure-Kinect-Sensor-SDK #883: cJSON parse error in Qt Application

avilleret, 2019-11-04:
**Describe the bug**

I'm trying to build a Qt application to record Azure Kinect Data.
As soon as I add `QApplication app(argc, argv)` in my `main(int argc, char** argv)`,  `k4a_device_open` fails at runtime with the following : 

```
Found 1 connected devices:
[2019-11-04 13:17:21.001] [error] [t=15677] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/calibration/calibration.c (391): get_camera_calibration(). cJSON_Parse error .5130159854888916,0.50180339813232422,0.49292910099029541,0.49299347400665283,6.3548245429992676,3.755314826965332,0.1709495335817337,6.6818685531616211,5.8674860000610352,0.96257597208023071,0,0,-6.2067297221801709E-6,1.9239363609813154E-5],"ModelType":"CALIBRATION_LensDistortionModelBrownConrady"},"Location":"CALIBRATION_CameraLocationD0","Purpose":"CALIBRATION_CameraPurposeDepth","MetricRadius":1.7399997711181641,"Rt":{"Rotation":[1,0,0,0,1,0,0,0,1],"Translation":[0,0,0]},"SensorHeight":1024,"SensorWidth":1024,"Shutter":"CALIBRATION_ShutterTypeUndefined","ThermalAdjustmentParams":{"Params":[0,0,0,0,0,0,0,0,0,0,0,0]}},{"Intrinsics":{"ModelParameterCount":14,"ModelParameters":[0.49859991669654846,0.50623816251754761,0.47083947062492371,0.6277928352355957,0.43399921059608459,-2.6028459072113037,1.54654860496521,0.31228992342948914,-2.4219636917114258,1.4691150188446045,0,0,0.00014593728701584041,-0.00028084038058295846],"ModelType":"CALIBRATION_LensDistortionModelB
[2019-11-04 13:17:21.001] [error] [t=15677] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (158): calibration_create(device->depthmcu, &device->calibration) returned failure in k4a_device_open()
0: Failed to open device
[2019-11-04 13:17:21.001] [error] [t=15677] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (53): k4a_device_t_get_context(). Invalid k4a_device_t (nil)
[2019-11-04 13:17:21.001] [error] [t=15677] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (955): Invalid argument to k4a_device_get_serialnum(). device_handle ((nil)) is not a valid handle of type k4a_device_t
0: Failed to get serial number length
[2019-11-04 13:17:21.001] [error] [t=15677] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (53): k4a_device_t_get_context(). Invalid k4a_device_t (nil)
[2019-11-04 13:17:21.001] [error] [t=15677] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (201): Invalid argument to k4a_device_close(). device_handle ((nil)) is not a valid handle of type k4a_device_t
```

Here is the complete code with a CMakeLists.txt :
[k4atest.zip](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/3803957/k4atest.zip)
The `k4atest.cpp` is a copy&paste from the `enumerate` sample with a `QApplication` call.

**To Reproduce**
Download the attached `k4atest.zip`, unzip and then  : 

    mkdir build-k4atest
    cd build-k4atest
    cmake ../k4atest
    make 
    ./k4atest

It output the error above.

**Expected output**

Edit the `k4atest.cpp`, comment out the `QApplication` call, rebuild then you get something like :

    Found 1 connected devices:
    0: Device "000083392212"

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 18.04
 - SDK Version: 1.3

avilleret, 2019-11-05:
I got the same error with commit 14c9372d4cd2420135dc5ace4c84552bbed4f34f 
But please note, that this doesn't happen if I run it inside `gdb`.

avilleret, 2019-11-05:
Same result with both gcc 7.4.0 and clang 8.0.0-3

avilleret, 2019-11-06:
Ok I got it.
Qt changes the locale in the `QApplication` constructor and that screws up `cJSON`.
Reverting the local to "C" with : 

        QLocale::setDefault(QLocale::C);
        setlocale(LC_ALL, "C");

Fix this annoying issue.

wes-b, 2019-11-06:
We will use this bug to make a change

PR #892 (2019-11-12): Set expected Locale for cJSON and calibration blob
## Fixes #883

### Description of the changes:
- Switching locale from process based to thread based and setting the locale based on the calibration raw data.
closed 2019-11-12 by wes-b