microsoft/Azure-Kinect-Sensor-SDK #884: Unable to compile with BUILD_SHARED_LIBS=ON

jasjuang, 2019-11-04:
**Describe the bug**
Compilation fails when cmake variable `BUILD_SHARED_LIBS` is set to `ON`

**To Reproduce**
```
git clone https://github.com/microsoft/Azure-Kinect-Sensor-SDK
cd Azure-Kinect-Sensor-SDK
mkdir build && cd build
cmake -DBUILD_SHARED_LIBS=ON ..
```

**Logs**
```
[ 60%] Built target k4a_queue
[ 60%] Built target k4a_deloader
Scanning dependencies of target k4a_dewrapper
Scanning dependencies of target k4a_tewrapper
[ 60%] Building C object src/tewrapper/CMakeFiles/k4a_tewrapper.dir/tewrapper.c.o
[ 60%] Building C object src/dewrapper/CMakeFiles/k4a_dewrapper.dir/dewrapper.c.o
../../src/logging/libk4a_logging.a(logging.cpp.o): In function `logger_init_once':
/home/ubuntu/ppa/Azure-Kinect-Sensor-SDK/src/logging/logging.cpp:139: undefined reference to `environment_get_variable'
/home/ubuntu/ppa/Azure-Kinect-Sensor-SDK/src/logging/logging.cpp:140: undefined reference to `environment_get_variable'
/home/ubuntu/ppa/Azure-Kinect-Sensor-SDK/src/logging/logging.cpp:141: undefined reference to `environment_get_variable'
collect2: error: ld returned 1 exit status
tools/deversion/CMakeFiles/deversion.dir/build.make:94: recipe for target 'bin/deversion' failed
make[2]: *** [bin/deversion] Error 1
CMakeFiles/Makefile2:6694: recipe for target 'tools/deversion/CMakeFiles/deversion.dir/all' failed
make[1]: *** [tools/deversion/CMakeFiles/deversion.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 60%] Linking CXX static library libk4a_usb_cmd.a
[ 60%] Built target k4a_usb_cmd
[ 60%] Linking CXX static library libk4a_tewrapper.a
[ 60%] Built target k4a_tewrapper
[ 61%] Linking CXX static library libk4a_dewrapper.a
[ 63%] Linking CXX shared library ../../../../../bin/libgtest.so
[ 63%] Built target k4a_dewrapper
[ 63%] Built target gtest
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
```

**Expected behavior**

I expect it compile successfully. Side note, it compiles successfully when `BUILD_SHARED_LIBS` is set to `OFF`.

**Desktop:**
 - OS with Version: Ubuntu 18.04
 - SDK Version:  Latest master

wes-b, 2019-11-05:
Thanks for the issue, we don't currently support that setting. Is there something you are trying to do that the SDK doesn't provide natively?

jasjuang, 2019-11-05:
@wes-b The automated script on our server has `BUILD_SHARED_LIBS=ON` when building for all external libraries and packaging them into `.deb` files for distribution. I can work around it by creating an exception for Azure Kinect SDK but almost all other open-source libraries can be built with `BUILD_SHARED_LIBS=ON` so I think it will be great if it can work for Azure Kinect SDK too.

baogege6661, 2019-11-13:
when i run "cmake ..-GNinja " this command, the following problems occurred
 CMake Error at CMakeLists.txt:23 (project):
  Running

   '/usr/sbin/ninja' '--version'

  failed with:

   die: error: must run as root

CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
please help me, thanks!!!

> @wes-b The automated script on our server has `BUILD_SHARED_LIBS=ON` when building for all external libraries and packaging them into `.deb` files for distribution. I can work around it by creating [...]

wes-b, 2019-11-13:
Looks like you didn't run as root, can you run `sudo cmake .. -GNinja`?

baogege6661, 2019-11-13:
First of all, thank you for your reply, but when I run this command on root, the cursor blinks and cannot proceed. I don't know what's wrong