microsoft/Azure-Kinect-Sensor-SDK #889: Gain color control value is kept in memory but image shows otherwise

lpasselin, 2019-11-05:
**Describe the bug**
Gain color control value is saved between two disconnects but the image shows otherwise. Getting a previously set gain value after a software restart or camera restart works fine (tested in k4aviewer and using C sdk). After a software restart, the captured image have a gain of 255. We can see this clearly when exposure is low (where gain variation can easily be seen in a color image).

**To Reproduce**
1. Open viewer, start camera stream.
2. Lower exposure to a value where changing the gain color control parameter can be noticed easily in the image.
3. Modify the gain. Set it to zero.
4. Stop camera.
5. Start camera. 
6. Notice gain value is still zero, yet image shows it is 255.

**Expected behavior**
After stopping cameras and restarting, gain parameter used to produce images should reflect the values obtained with get_color_control.

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
1. Tested on Windows 10 (surface device) with SDK version 1.2 and 1.3
2. Tested on Arch Linux with SDK version 1.2

**Additional context**

wes-b, 2019-11-06:
Firmware team has this issue in their queue.

lpasselin, 2019-12-11:
Is there anyway for me to help with this issue?

wes-b, 2019-12-11:
Thanks, we have everything we need to fix this. We currently have some higher priority FW fixes in work so this fix will take some time. You should not be blocked though as you can write the value to a know setting before starting the camera.

lpasselin, 2019-12-11:
> You should not be blocked though as you can write the value to a know setting before starting the camera.

This does not work.
A gain setting that is set before starting the camera is not used by the camera.

wes-b, 2019-12-11:
Gotcha, you will need FW 1.6.107078014 or better. This version will remove the fixed gain that is being applied and allows the gain to be set.

lpasselin, 2019-12-11:
Great! Is this firmware available publicly yet? I can't find anywhere AzureKinectDK_Fw_1.6.107078014.bin

wes-b, 2019-12-11:
Correct that it is not yet released.

lpasselin, 2020-02-27:
Fixed with device software update. Thanks!