microsoft/Azure-Kinect-Sensor-SDK #890: Please add the CMake file for offline_processor

legacydev, 2019-11-06:
The source does not include the CMake file for the offline_processor. Can this be added?

wes-b, 2019-11-06:
Our goal with #864 was to get enough of a CMake system in place to enable the community to help build out the samples. 

We would be happy to answer questions if issues are encountered.

legacydev, 2019-11-07:
I've added a comment to #864 as I am unable to build it. Currently haven't had a chance to go through moving this particular project to CMake, yet.

KankshaZ, 2020-01-29:
The following steps helped me build offline_processor:
1. Added [json.hpp](https://github.com/nlohmann/json/blob/develop/single_include/nlohmann/json.hpp) to sample_helper_includes folder
2. Appended the line `add_subdirectory(offline_processor)` to CMakeLists.txt in body-tracking-samples
3. Added CMakeLists.txt to offline_processer which contains
```
add_executable(offline_processor main.cpp)
target_include_directories(offline_processor PRIVATE ../sample_helper_includes)
target_link_libraries(offline_processor PRIVATE 
    k4a
    k4abt
    k4arecord
    window_controller_3d::window_controller_3d
    glfw::glfw
    )

qm13, 2022-05-17:
Fixed

closed 2022-05-17 by qm13