microsoft/Azure-Kinect-Sensor-SDK #893: There is something wrong with the C++ interface of depth_image_to_color_camera?

nonlinear1, 2019-11-07:
I write a example of C++ interface of SDK, I found that I cann't use the "depth_image_to_color_camera" of C++ version. when program run to     "transformation.depth_image_to_color_camera(depth_image, &transformed_depth_image)", the program is crashed and and give me the following errors:
[2019-11-07 07:59:37.385] [error] [t=3131] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/image/image.c (51): k4a_image_t_get_context(). Invalid k4a_image_t (nil)
[2019-11-07 07:59:37.385] [error] [t=3131] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/image/image.c (390): Invalid argument to image_get_width_pixels(). image_handle ((nil)) is not a valid handle of type k4a_image_t
[2019-11-07 07:59:37.385] [error] [t=3131] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/image/image.c (51): k4a_image_t_get_context(). Invalid k4a_image_t (nil)
[2019-11-07 07:59:37.385] [error] [t=3131] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/image/image.c (397): Invalid argument to image_get_height_pixels(). image_handle ((nil)) is not a valid handle of type k4a_image_t
[2019-11-07 07:59:37.385] [error] [t=3131] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/image/image.c (51): k4a_image_t_get_context(). Invalid k4a_image_t (nil)
[2019-11-07 07:59:37.385] [error] [t=3131] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/image/image.c (404): Invalid argument to image_get_stride_bytes(). image_handle ((nil)) is not a valid handle of type k4a_image_t
[2019-11-07 07:59:37.385] [error] [t=3131] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/image/image.c (51): k4a_image_t_get_context(). Invalid k4a_image_t (nil)
[2019-11-07 07:59:37.385] [error] [t=3131] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/image/image.c (383): Invalid argument to image_get_format(). image_handle ((nil)) is not a valid handle of type k4a_image_t
[2019-11-07 07:59:37.385] [error] [t=3131] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/image/image.c (51): k4a_image_t_get_context(). Invalid k4a_image_t (nil)
[2019-11-07 07:59:37.385] [error] [t=3131] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/image/image.c (359): Invalid argument to image_get_buffer(). image_handle ((nil)) is not a valid handle of type k4a_image_t
[2019-11-07 07:59:37.385] [error] [t=3131] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/transformation/rgbz.c (680): transformation_depth_image_to_color_camera_validate_parameters(). Transformed depth image data is null.
[2019-11-07 07:59:37.385] [error] [t=3131] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/transformation/transformation.c (710): transformation_depth_image_to_color_camera_validate_parameters( &transformation_context->calibration, &transformation_context->depth_camera_xy_tables, depth_image_data, depth_image_descriptor, custom_image_data, custom_image_descriptor, transformed_depth_image_data, transformed_depth_image_descriptor, transformed_custom_image_data, transformed_custom_image_descriptor) returned failure in transformation_depth_image_to_color_camera_custom()
[2019-11-07 07:59:37.385] [error] [t=3131] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (1197): transformation_depth_image_to_color_camera_custom(transformation_handle, depth_image_buffer, &depth_image_descriptor, custom_image_buffer, &dummy_descriptor, transformed_depth_image_buffer, &transformed_depth_image_descriptor, transformed_custom_image_buffer, &dummy_descriptor, interpolation_type, invalid_custom_value) returned failure in k4a_transformation_depth_image_to_color_camera()
terminate called after throwing an instance of 'k4a::error'
  what():  Failed to convert depth map to color camera geometry!
Aborted (core dumped)

In fact, depth_image and color_image  have images. and c version of this function is OK in my environment!
 the following is my environment and codes:
environment:
hardware:
cpu i7
ram 16G
GPU nvidia gtx 1070

the software enviroment:
os: ubuntu18.04
sdk1.3.0
test_k4a.cpp:
#include <iostream>
#include <k4a/k4a.hpp>

using namespace std;

int main()
{  
    k4a::device m_device;
    m_device = k4a::device::open(0);
    std::cout << m_device.get_serialnum()<< std::endl;
    k4a::transformation transformation;
    k4a::capture capture ;
    k4a::image depth_image;
    k4a::image color_image ;
    k4a::image transformed_depth_image;

    k4a_device_configuration_t deviceConfig;

    deviceConfig.synchronized_images_only = true;
    deviceConfig.color_format = K4A_IMAGE_FORMAT_COLOR_BGRA32;
    deviceConfig.color_resolution = K4A_COLOR_RESOLUTION_720P;
    deviceConfig.depth_mode = K4A_DEPTH_MODE_NFOV_UNBINNED;
    deviceConfig.camera_fps = K4A_FRAMES_PER_SECOND_30;
    deviceConfig.wired_sync_mode = K4A_WIRED_SYNC_MODE_STANDALONE;
    deviceConfig.subordinate_delay_off_master_usec = 0;
    deviceConfig.subordinate_delay_off_master_usec = 0;
    deviceConfig.disable_streaming_indicator = false;
  
    k4a::calibration calibration;
    calibration = m_device.get_calibration(K4A_DEPTH_MODE_NFOV_UNBINNED, K4A_COLOR_RESOLUTION_720P);
    transformation = k4a::transformation(calibration);
    m_device.start_cameras(&deviceConfig);

    std::chrono::milliseconds CameraPollingTimeout(0);
    m_device.get_capture(&capture,CameraPollingTimeout);
    std::cout << "This is test!" << std::endl;

    depth_image = capture.get_depth_image();
    if(!depth_image)
    {
       std::cout << "No depth_image!" << std::endl;
    }
    color_image = capture.get_color_image();
    if(!color_image)
    {

    std::cout << "No color_image!" << std::endl;
   
    }

    transformation.depth_image_to_color_camera(depth_image, &transformed_depth_image);
  
    std::cout << "This is the end" << std::endl;

}
CMakeLists.txt
cmake_minimum_required(VERSION 3.9.0)
find_package(k4a)
add_executable(test_k4a 
    test_k4a.cpp
    )

target_link_libraries(test_k4a     
k4a::k4a
    )
Could someone know why?
Thank you very much!

RoseFlunder, 2019-11-07:
You need to create your "transformed_depth_image" before you pass the reference to the transformation method like this:
```
k4a::image transformed_depth_image = k4a::image::create(
            K4A_IMAGE_FORMAT_DEPTH16,
            getColorWidth(),
            getColorHeight(),
            getColorWidth() * (int)sizeof(uint16_t));
```

nonlinear1, 2019-11-07:
OK，I will try ,thank you very much

closed 2019-11-08 by wes-b