microsoft/Azure-Kinect-Sensor-SDK #894: Cannot reference Sensor SDK from .NET Class Library

dotMorten, 2019-11-07:
**Describe the bug**
Create a new .NET Standard Class Library, and you'll be prevented from referencing the SDK due to the class library being AnyCPU.
This limitation should not be put on libraries but only on applications.

**To Reproduce**
1. Create a new .NET Standard Class Library
2. Add the Microsoft.Azure.Kinect.Sensor nuget package
3. Notice the error in the output window:

```
1>C:\Users\mn\.nuget\packages\microsoft.azure.kinect.sensor\1.3.0\build\netstandard2.0\Microsoft.Azure.Kinect.Sensor.targets(4,5): error : Azure Kinect only supports the x86/x64 platform ('AnyCPU' not supported)
1>Done building project "Microsoft.Azure.Kinect.BodyTracking.csproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========
```

**Expected behavior**
Check should only be done on applications, not libraries.

**Side note**
Since such a class library would transitively bring in the kinect sensor package, you should also add the same build files in the nuget package to /buildTransitive/ to ensure the .targets file are run on applications not directly referencing that nuget package.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS with Version: N/A
 - SDK Version:  1.3.0
 - Firmware version:  N/A

**Additional context**

dotMorten, 2019-11-07:
In addition referencing the project from a .NET Core WPF app, you'll hit a similar issue because it doesn't use the $(Platform) value, but relies on runtime identifiers instead.

Brent-A, 2019-11-07:
Thanks. We should be publishing a new alpha after we complete #822 and I'll try integrating those suggestions to the NuGet release.

FYI, #822 adds some APIs to expose raw handle access that will make it possible to interop with the body tracking SDK.

vpenades, 2020-03-10:
Is this still an issue? I thought it would be fixed for 1.4.0...

dotMorten, 2020-07-11:
@qm13 Was this addressed?

qm13, 2020-07-14:
@dotMorten my bad. I believe this is still an issue.

tmoody460, 2020-07-17:
Any chance there's a workaround for this until this can get fixed?