microsoft/Azure-Kinect-Sensor-SDK #896: Building Azure Kinect Samples via cmake fails

legacydev, 2019-11-08:
Trying to build Azure Kinect Samples on Ubuntu 18.04 but unable to follow build instructions

Steps:

1. cloned repo https://github.com/microsoft/Azure-Kinect-Samples
2. followed instructions on this link: https://docs.microsoft.com/en-us/windows-server/administration/linux-package-repository-for-microsoft-software
3. ```apt install libk4abt0.9-dev```
4. ```apt install libxi-dev```
5. installed vcpckg
6. installed libigl[glfw] (and glfw3)
7. Changed to git project root
8. ```mkdir build```
9. ```cd build```
10. ```cmake .. -GNinja```

This is the error:
```CMake Error at body-tracking-samples/extern/glfw/CMakeLists.txt:16 (add_subdirectory):
  The source directory
    /home/user/Documents/Azure-Kinect-Samples/body-tracking-samples/extern/glfw/src
  does not contain a CMakeLists.txt file.
CMake Error at body-tracking-samples/extern/glfw/CMakeLists.txt:18 (add_library):
  add_library cannot create ALIAS target "glfw::glfw" because target "glfw"
  does not already exist.
-- Configuring incomplete, errors occurred!
```

Not sure if I missed a step somewhere or if this is a build issue. Would appreciate any assistance!

legacydev, 2019-11-08:
Whoops - missed downloading the submodule. Building now, thanks!

closed 2019-11-08 by legacydev

emmanuel-senft, 2020-01-16:
Hi, I've got the same issue as you, could you please specify which submodule you downloaded?

KankshaZ, 2020-01-29:
A simple git clone is not enough to download the submodules. After cloning the repository, use the following command
`git submodule update --init --recursive`
This should fix your issue!

AswinkarthikeyenAK, 2020-02-28:
Hi, Quick question, I had the same issue but it got fixed after installing the submodule. But I not able to execute the file after CMake. 
[33/33] Linking CXX executable bin/simple_3d_viewer

Therefore, now I am not able to execute the file for body tracking. I am using Ubuntu 18.04. 

Can someone provide a solution if possible?

KankshaZ, 2020-03-02:
Have you tried running it with sudo?

AswinkarthikeyenAK, 2020-03-02:
Yes, I did.
I am running into another problem now.
Shader Error: 0:1(10): error: GLSL 4.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

madelinegannon, 2020-06-17:
> Hi, Quick question, I had the same issue but it got fixed after installing the submodule. But I not able to execute the file after CMake.
> [33/33] Linking CXX executable bin/simple_3d_viewer [...]

How did you resolve this? Having the same cmake issue on Ubuntu 18.04

romilpunetha, 2021-03-29:
Running the ninja command gives me the following error:
```
1/1] Linking CXX executable bin/simple_3d_viewer
FAILED: bin/simple_3d_viewer 
: && /usr/bin/c++    -Wl,--as-needed,-rpath-link=/home/romil/Azure-Kinect-Samples/build/bin body-tracking-samples/simple_3d_viewer/CMakeFiles/simple_3d_viewer.dir/main.cpp.o  -o bin/simple_3d_viewer  -Wl,-rpath,"\$ORIGIN"  -lk4a  -lk4abt  body-tracking-samples/sample_helper_libs/window_controller_3d/libwindow_controller_3d.a  body-tracking-samples/extern/glfw/src/src/libglfw3.a  /usr/lib/x86_64-linux-gnu/librt.so  -lm  -ldl  /usr/lib/x86_64-linux-gnu/libX11.so  -lpthread  /usr/lib/x86_64-linux-gnu/libXrandr.so  /usr/lib/x86_64-linux-gnu/libXinerama.so  /usr/lib/x86_64-linux-gnu/libXcursor.so && :
/usr/bin/ld: body-tracking-samples/simple_3d_viewer/CMakeFiles/simple_3d_viewer.dir/main.cpp.o: in function `PlayFile(InputSettings)':
main.cpp:(.text+0xd42): undefined reference to `k4a_playback_open'
/usr/bin/ld: main.cpp:(.text+0xd89): undefined reference to `k4a_playback_get_calibration'
/usr/bin/ld: main.cpp:(.text+0xefd): undefined reference to `k4a_playback_get_next_capture'
/usr/bin/ld: main.cpp:(.text+0x10df): undefined reference to `k4a_playback_close'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

```

ma-nu-ell, 2021-11-14:
This could be resolved by adding k4arecord in target_link_libraries in CMakeLists.txt of simple_3d_viewer.
Thanx to [haeginh](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1296#issue-666281059)

AswinkarthikeyenAK, 2022-01-22:
I have a issue while running simple_3d_viewer in ubuntu 
`libk4a.so.1.4: cannot open shared object file: No such file or directory`
Can someone please let me know what needs to be done to fix this?

Thanks

ma-nu-ell, 2022-01-23:
Here is a good instruction how to run it on ubuntu 18.4https://vinesmsuic.github.io/2020/10/31/azure-kinect-setup/#invoke-k4aviewer-without-rootThx 2 Vine