microsoft/Azure-Kinect-Sensor-SDK #898: Manual ISO speed control

catid, 2019-11-09:
Please provide a control for ISO speed of the cameras.  Currently we are able to read the ISO speed metadata for each frame but cannot adjust this setting.  This prevents users from configuring all of the cameras in a multi-camera rig the same way to capture consistent lighting.

wes-b, 2019-11-11:
What about [k4a_image_get_iso_speed](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_ga3626cef655d79190d2e002c3c4030872.html) or [k4a_image_set_iso_speed](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_ga5a64f64515ff4dff317fec92a8427f45.html#ga5a64f64515ff4dff317fec92a8427f45)

catid, 2019-11-12:
This does not change camera settings, right?

wes-b, 2019-11-12:
Gotcha. No they don't. This sounds like a good idea on the surface, but I wonder if an ISO setting on one camera is really the same as it is on another?

catid, 2019-11-13:
It's not easy for me to answer that one.  I don't have a way currently to test without modifying the SDK.  One way would be to provide the libuvc handle up through the API so the user can attempt to manage this manually, but I'm not sure how that works with the camera firmware and could fight with the SDK API.  And even then it's possible no one will use the feature.

wes-b, 2019-12-05:
Closing this for now, open to revisit if more feedback is recieved.

closed 2019-12-05 by wes-b