microsoft/Azure-Kinect-Sensor-SDK #900: Multiple Cameras and Body Tracking

legacydev, 2019-11-11:
Trying to stitch point cloud data together and run it through the body tracking. However, this doesn't seem possible - Looking at the [green screen example](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/green_screen), this creates an open cv matrix and renders it. However, creating a k4a_capture_t will only take a single depth and color image.

How can I create a k4a_capture_t with point clouds (or even depth images)?

wes-b, 2019-11-11:
What do you want to do with the k4a_capture_t? You can assign any image type with [k4a_capture_set_color_image](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_ga28824350a175d8eeba7c7a7d2753c2cf.html), [k4a_capture_set_depth_image](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_gad28b520b45096dd430d2a14b6052ebae.html#gad28b520b45096dd430d2a14b6052ebae), and [k4a_capture_set_ir_image](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_ga473abb7de733bf5f53329f3b1a86594c.html#ga473abb7de733bf5f53329f3b1a86594c). There is not validation that the image cormat matches color, depth or ir.

legacydev, 2019-11-11:
I've got 2 captures from 2 kinect cameras and I want to translate the point cloud of the second camera to align to the first. That works fine, but I can't add the result to the second camera to the first and then run it through the body tracking.

With this:
k4a_playback_get_next_capture(playback_handle, &capture_handle);

I assume I need to build out the capture with both point cloud sources. I'm not really sure how to do that.

yijiew, 2019-11-11:
Hi @legacydev : the body tracking doesn't support tracking bodies for stitched point cloud data. You can only run body tracking for a single capture at a time. In the next release, we are going to allow the user to create multiple trackers. Then for captures from different cameras, you can create multiple trackers to track them separately and come up with your own smart way to fuse the results together.

sopangandhe, 2019-11-12:
Hi @yijew, what is the timeline on the release of the next update?

qm13, 2019-11-12:
We expect to release an update with this functionality by EOY.

kklouzal, 2019-11-14:
> Hi @legacydev : ...**come up with your own smart way to fuse the results together**...
Isn't one of the major selling points for this new Azure Kinect that you can use multiple sensors to solve complex problems? Saying the user ultimately has to 'figure it out on their own' defeating the purpose? It seems appropriate to me that body tracking, a highly sought after feature, using multiple sensors, would be the sort of functionality which is provided out of the box.

legacydev, 2019-11-15:
I believe the thought is to filter joints with less confidence - assuming that a different camera may have a better joint view. This is different from what I was looking to do - which is build a more detailed point cloud that can be processed for a better "view" of the space

billpottle, 2019-11-19:
It would also be amazing if the multiple cameras could have their capture times shifted by half of a frame time. That would allow something like 60 FPS, rather than the current max of 30. This would be a game changer for analyzing athletic movements.

qm13, 2019-11-19:
Technically you can hardware synchronize two Azure Kinect cameras and impose a 1/2 frame delay between the two cameras. If you intention is to interleave the frames from the two cameras you would need to translate them due to the physical offsets of the two cameras (since the two cameras will not be on the same axis).

qm13, 2019-12-07:
Fixed in v0.9.5.

closed 2019-12-07 by qm13