microsoft/Azure-Kinect-Sensor-SDK #902: All VersionInfo.cs files are missing

amang2205, 2019-11-13:
None of the C# projects can be built. The VersionInfo.cs file is missing for every single project, and a bunch of .dlls are also missing: depthengine_2_0.dll, k4a.dll, k4a.pdb...

tesych, 2019-11-13:
@amang2205 Maybe you forgot this part?
The C# build depends on files generated by the CMake and Ninja build systems. Therefore the CMake configuration steps generates files that will be consumed by the C# Build and Test systems to find the needed components. Therefore, you need to run the CMake configuration and build before loading the C# solution.
The VersionInfo.cs should be generated by this step.

amang2205, 2019-11-14:
I definitely didn't run the CMake build, but that explains it clearly.
In my opinion though, it's regretful that I can't have a proper F5 experience here. I'd expect when I want to run a sample to run a sample, not have to manually copy over files from the SDK (namely the .dlls), do a CMake build etc. It really spoils the experience...

closed 2019-11-14 by tesych