microsoft/Azure-Kinect-Sensor-SDK #903: Automatic White Balance out of range

catid, 2019-11-13:
The AWB feature, when set to automatic, is able to select WhiteBalanceUsec values that are below the minimum of the allowed range, based on read-back.  For example indoors I am seeing WhiteBalanceUsec around 2000, while the lowest I can manually set is 2500.  This means that I am unable to enable AWB, and then "lock" the setting so it does not change.

Motivation: Locking seems to be useful for matching colors between cameras.  The best approach I've come up with so far involves first starting all the cameras in AWB/AE mode and then locking AE and performing an iterative algorithm to normalize overlapping voxels.  However AWB cannot be locked, so it's hard to be 100% sure that the camera settings aren't changing during the tuning.  And later on during capture the AWB might fluctuate for different cameras in response to scene lighting changes.

catid, 2019-11-16:
Here's a motivating example:

[image]

The cameras that see the (mainly blue) screen are turning the rest of the picture yellow to compensate.

The AWB read-back shows: 2076, 2058, 2270, 2394 (two cameras see the screen, two do not).

Manually configuring the cameras with AWB = 2000 fails (API returns an invalid input error).

Manually configuring the cameras with AWB = 2500 succeeds, but then the scene looks yellow.

wes-b, 2019-12-05:
#793

wes-b, 2019-12-18:
I can also reproduce this. We will need a firmware change to address this.

wes-b, 2020-01-24:
The camera experts I have chatted with have said that it is not uncommon for auto camera modes to achieve settings that are not reachable via manual settings. As a result we don't plan to address this. 

We will reconsider if there is enough feedback, please open a request at https://aka.ms/azurekinectfeedback, where the community can vote on this issue and others.

closed 2020-01-24 by wes-b