microsoft/Azure-Kinect-Sensor-SDK #907: Error executing example "Build an Azure Kinect body tracking application"

choonyip, 2019-11-15:
https://docs.microsoft.com/ja-jp/azure/Kinect-dk/build-first-body-app
I followed the example showed in the page above, on Ubuntu 18.04,
with source code https://github.com/choonyip/bodytracking/blob/master/bodytracking/main.c
and CMakeLists.txt https://github.com/choonyip/bodytracking/blob/master/bodytracking/CMakeLists.txt
But I am getting error:

```
/onnxruntime_src/onnxruntime/core/providers/cuda/cuda_call.cc:97 bool 
onnxruntime::CudaCall(ERRTYPE, const char*, const char*, ERRTYPE, const char*) [with ERRTYPE = 
cudaError; bool THRW = true] /onnxruntime_src/onnxruntime/core/providers/cuda/cuda_call.cc:91 
bool onnxruntime::CudaCall(ERRTYPE, const char*, const char*, ERRTYPE, const char*) [with ERRTYPE 
= cudaError; bool THRW = true] CUDA failure 35: CUDA driver version is insufficient for CUDA 
runtime version ; GPU=0 ; hostname=cy8182-TM1604 ; expr=cudaSetDevice(device_id_);
Stacktrace:

Stacktrace:

```

RoseFlunder, 2019-11-15:
According to the error message your graphics card driver is insufficient.
Did you select the newest proprietary NVidia driver in the "Additional drivers" tab in the Ubuntu Software Settings?

choonyip, 2019-11-15:
@RoseFlunder Thank you for your reply. I changed to the newest proprietary NVidia driver, and it worked! Really appreciate your help!

closed 2019-11-15 by wes-b