microsoft/Azure-Kinect-Sensor-SDK #908: i am making changes to the SDK, i have built my own copy of the SDK from source. Follow the instruction in building for how to build from source,

1216704928, 2019-11-15:
**Describe the bug**

I am making changes to the SDK, i have built my own copy of the SDK from source,and fellow the instruction in building for how to build from source.After compiling, run the generated k4aviewer. exe. Although the interface window is opened, it cannot "start".
**To Reproduce**

[image]
[image]
[image]
http://ww1.sinaimg.cn/large/005Zn2sIgy1g8yoyqc9bdj30oa02bq2u.jpg
http://ww1.sinaimg.cn/large/005Zn2sIgy1g8yoy90mz3j30pe0b3dg9.jpg
http://ww1.sinaimg.cn/large/005Zn2sIgy1g8yozqxy9mj313u0oygmp.jpg
**Expected behavior**

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: 
 - SDK Version:  
 - Firmware version:  

**Additional context**

wes-b, 2019-11-15:
The viewer has the option of collecting logs, can you share them?

denkywu, 2019-11-26:
I have the same problem.

[image]

denkywu, 2019-11-27:
> I have the same problem.
> 
> [image]

My problem solved by updating my graphics drivers (NVIDIA) instead of integrated graphics.

closed 2019-11-27 by wes-b