microsoft/Azure-Kinect-Sensor-SDK #914: What is metric_radius in k4a_calibration_intrinsic_parameters_t?

jasjuang, 2019-11-15:
I am wondering what exactly is [metric_radius](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/develop/structk4a__calibration__intrinsic__parameters__t_1_1__param.html#a91b39ee152b891ae79641ad8e6137adb) that is part of the intrinsic parameters (k4a_calibration_intrinsic_parameters_t)? I can understand all the other parameters but is having trouble understanding this one.

catid, 2019-11-16:
Imagine each camera captures roughly a cone in space - This is the radius of that cone.  The cone is slightly larger than the actual volume that is captured.

jasjuang, 2019-11-17:
@catid thanks for the explanation, my next question is when I read the metric_radius parameter, it shows the value 0. Is that expected? Based on your explanation I would imagine metric_radius to be a non zero number in millimeters.

catid, 2019-11-17:
Searching through the code I see some hints like this:
        // metric_radius equals 0 means that calibration failed to estimate this parameter
        if (cal->metric_radius <= 0.0001f)
        {
            cal->metric_radius = 1.7f; // default value corresponds to ~120 degree FoV
        }

jasjuang, 2019-11-17:
@catid do I have put place the azure kinect in a certain way for it self-calibrate? I am under the impression that the intrinsic parameters come with the device (i.e. factory settings).

catid, 2019-11-17:
Yeah the calibration data is loaded from the device itself.  Not sure why yours is zero.  On my unit for NFOV2X2BINNED mode (320x288 depth image) the metric_radius = 1.74 and for the RGB color camera it's 1.70.

Maybe none of the units have this configured?

catid, 2019-11-17:
I set breakpoints in all the checked where it uses the metric_radius and it is currently unused at least in the modes I'm testing.

catid, 2019-11-17:
Checked narrow FOV mode and wide FOV mode: It seems to be unused in both modes.  Disabling all checks related to it does not affect the result in any way.

jasjuang, 2019-11-17:
I was shooting on narrow FOV, 640x576 depth image, and 3840x2160 color image. Are you able to get a non zero metric_radius for this setup?

closed 2020-01-10 by tesych