microsoft/Azure-Kinect-Sensor-SDK #916: Add/use the k4abt library path to search for dnn_model_2_0.onnx

conatsera, 2019-11-17:
**Is your feature request related to a problem? Please describe.**
External applications that dynamically link libraries that use K4ABT will have the external project's root searched for the DNN model. For example, in a SteamVR device driver, the K4ABT library will search `Steam\steamapps\common\SteamVR`.

**Describe the solution you'd like**
Have the library use it's own path as the hint to locating the DNN model.

**Describe alternatives you've considered**
Using an installer to place the model in the searched directory

**Additional context**
A SteamVR driver:
https://github.com/conatsera/K4A_OpenVR

yijiew, 2019-11-19:
Hi @conatsera , thanks for your feedback! We are going to take a look to see how we could solve this issue. 
Could you clarify a bit about what your first suggested solution? 
> Have the library use it's own path as the hint to locating the DNN model.
Do you mean search the dnn_model_2_0.onnx file at the k4abt.dll path?

conatsera, 2019-11-19:
Yes. For example calling `GetModuleFileName(dllHandle, path, pathSize)` on windows.

qm13, 2019-12-07:
Fixed in v0.9.5.

closed 2019-12-07 by qm13