microsoft/Azure-Kinect-Sensor-SDK #917: How to get FOV/FOI using Azure Kinect Sensor SDK?

UnaNancyOwen, 2019-11-18:
Currently, I can't find how to get FOV/FOI with Azure Kinect Sensor SDK.
Is there a way to get these values that depend on the sensor mode?

* Color camera FOV
https://docs.microsoft.com/en-us/azure/kinect-dk/hardware-specification#color-camera-supported-operating-modes
  
  | Mode | FOI | 
  | ------ | ---- |
  | 2160P, 1440P, 1080P, 720P | 90°x59° |
  | 3072P, 1536P | 90°x74.3° |

* Depth camera FOI
https://docs.microsoft.com/en-us/azure/kinect-dk/hardware-specification#depth-camera-supported-operating-modes
  
  | Mode | FOI | 
  | ------ | ---- |
  | NFOV | 75°x65° |
  | WFOV | 120°x120° |

**Describe the solution you'd like**

Add FOV/FOI to <code>k4a_calibration_t</code>.
The API (<code>k4a_device_get_calibration()</code>) to get it receives color_resolution and depth_mode.
FOV/FOI values ​​depend only on color_resolution and depth_mode. We can be implemented quickly.

UnaNancyOwen, 2019-11-18:
* include/k4a/k4atypes.h
```patch
 typedef struct _k4a_calibration_camera_t
     int resolution_width;                    /**< Resolution width of the calibration sensor. */
     int resolution_height;                   /**< Resolution height of the calibration sensor. */
     float metric_radius;                     /**< Max FOV of the camera. */
+    float fov_holizontal;                    /**< FOV holizontal of the camera. */
+    float fov_vertical;                      /**< FOV vertical of the camera. */
 } k4a_calibration_camera_t;
```
* src/transformation/mode_specific_calibration.c
```patch
 k4a_result_t
 transformation_get_mode_specific_depth_camera_calibration(const k4a_calibration_camera_t *raw_camera_calibration,
                                                         const k4a_depth_mode_t depth_mode,
                                                         k4a_calibration_camera_t *mode_specific_camera_calibration)
 {

     /* ... */

     switch (depth_mode)
     {
     case K4A_DEPTH_MODE_NFOV_2X2BINNED:
     {
+        mode_specific_camera_calibration->fov_holizontal = 75.0f;
+        mode_specific_camera_calibration->fov_vertical = 65.0f;
         mode_specific_camera_calibration.
         k4a_camera_calibration_mode_info_t mode_info = { { 512, 512 }, { 96, 90 }, { 320, 288 } };
         return TRACE_CALL(
             transformation_get_mode_specific_camera_calibration(raw_camera_calibration,
                                                                 &mode_info,
                                                                 mode_specific_camera_calibration,
                                                                 /* pixelized_zero_centered_output = */ true));
    }

    /* ... */
```
```patch
 k4a_result_t
 transformation_get_mode_specific_depth_camera_calibration(const k4a_calibration_camera_t *raw_camera_calibration,
                                                         const k4a_depth_mode_t depth_mode,
                                                         k4a_calibration_camera_t *mode_specific_camera_calibration)
 {

     /* ... */

     switch (color_resolution)
     {
     case K4A_COLOR_RESOLUTION_720P:
     {
+        mode_specific_camera_calibration->fov_holizontal = 90.0f;
+        mode_specific_camera_calibration->fov_vertical = 59.0f;
         k4a_camera_calibration_mode_info_t mode_info = { { 1280, 960 }, { 0, 120 }, { 1280, 720 } };
         return TRACE_CALL(
             transformation_get_mode_specific_camera_calibration(mode_specific_camera_calibration,
                                                                 &mode_info,
                                                                 mode_specific_camera_calibration,
                                                                 /* pixelized_zero_centered_output = */ true));
     }

     /* ... */
```

wes-b, 2019-11-18:
Thanks for the suggestion. Does this lack of information block any development? Our interpretation of this request is that this would be nice to have but there is nothing blocking due to the info being available here: https://docs.microsoft.com/en-us/azure/Kinect-dk/hardware-specification#depth-camera-supported-operating-modes

UnaNancyOwen, 2019-11-19:
Yes. Currently, It has hardcoded in my project because there is nothing we can do.
Hoever, that information was available in Kinect SDK v2 at least. 
(<code>IFrameDescription::get_HorizontalFieldOfView()</code>,<code>IFrameDescription::get_VerticalFieldOfView()</code>)
I think the Azure Kinect SDK should also provide it.