microsoft/Azure-Kinect-Sensor-SDK #918: Official OpenGL recommendation for Ubuntu 18.04

scimad, 2019-11-18:
I installed `libk4a1.3`, `libk4a1.3-dev`, `k4a-tools`, and updated the udev rules as mentioned in documentation [here]

I am getting the following error while running the `k4aviewer` (both with or without sudo) though everything is perfect in while running k4aviewer.exe on Windows:

[image]
I get this message in the terminal:

```
Error: OpenGL 4.4 not supported. Please install latest graphics drivers.
[2019-11-18 09:54:21.190] [error] [t=5834] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (154): depth_engine_start_helper(). Depth engine create and initialize failed with error code: 207.
[2019-11-18 09:54:21.190] [error] [t=5834] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (157): depth_engine_start_helper(). OpenGL 4.4 context creation failed. You could try updating your graphics drivers.
[2019-11-18 09:54:21.190] [error] [t=5834] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (160): deresult == K4A_DEPTH_ENGINE_RESULT_SUCCEEDED returned failure in depth_engine_start_helper()
[2019-11-18 09:54:21.190] [error] [t=5834] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (194): depth_engine_start_helper(dewrapper, dewrapper->fps, dewrapper->depth_mode, &depth_engine_max_compute_time_ms, &depth_engine_output_buffer_size) returned failure in depth_engine_thread()
[2019-11-18 09:54:21.190] [error] [t=5796] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (552): dewrapper_start(). Depth Engine thread failed to start
[2019-11-18 09:54:21.190] [error] [t=5796] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth/depth.c (451): dewrapper_start(depth->dewrapper, config, depth->calibration_memory, depth->calibration_memory_size) returned failure in depth_start()
[2019-11-18 09:54:21.190] [error] [t=5796] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (359): cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming()
[2019-11-18 09:54:21.190] [error] [t=5796] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (362): depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063
[2019-11-18 09:54:21.190] [error] [t=5796] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (891): depth_start(device->depth, config) returned failure in k4a_device_start_cameras()

```

And I don't get the IR, color, depth, IMU data but I am getting microphone data.

My system has the following configurations:
OS: Ubuntu 18.04
OpenGL version string: 4.3 (Compatibility Profile) Mesa 19.0.8

Also tried installing other versions of `libk4a` and `libk4a-dev` but no luck.

Is it caused by OpenGL version mismatch? If yes, what is the recommendation and how to proceed if OpenGL isn't the issue?

[here]: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md

scimad, 2019-11-18:
Further information:
After rechecking, the issue is found to be thrown only when `Enable Depth Camera` is checked and `Depth mode` is set to anything other than `Passive IR`.

wes-b, 2019-11-18:
You need to update your video driver, I think OpenGL 4.4 is our min version and that 1st error message needs to be fixed.

wes-b, 2019-11-18:
The first error I believe is from OpenGL because we are attempting to initialize for 4.4.
`Error: OpenGL 4.4 not supported. Please install latest graphics drivers.`
The next 2 error messages confirm this, so yes, please update your driver.
```
dewrapper.c (154): depth_engine_start_helper(). Depth engine create and initialize failed with error code: 207.
dewrapper.c (157): depth_engine_start_helper(). OpenGL 4.4 context creation failed. You could try updating your graphics drivers.
```

scimad, 2019-11-19:
I have update the drivers for my graphics card and the OpenGL is upgraded to 4.6. Now the issue is no more. Thanks.

However, after the installation of new drivers I am getting 'Failed to initialize audio backend: incompatible device!' 
[image]

And my microphone reading doesn't show up. (which was previously showing up even with no RGBD data showing up). [Though my issue is resolved, I am giving reporting this issue for info only (my application doesn't need audio stream)].

closed 2019-11-19 by scimad

jackneer, 2020-01-07:
@scimad I got the same error in the same env. as yours.  Please share how you solve the problem or any hints could be very helpful. Thanks!

scimad, 2020-01-09:
@jackneer, I installed nvidia graphics drivers using `Software & Updates` >> `Additional Drivers` >> `Using NVIDIA driver metapackage ...` apparently which updated OpenGL on my ubuntu system if that helps you.

Apollo19971206, 2020-07-27:
> @jackneer，我使用`Software & Updates`>> `Additional Drivers`>> 安装了nvidia图形驱动程序，`Using NVIDIA driver metapackage ...`显然可以在我的ubuntu系统上更新OpenGL，如果有帮助的话。

Hello.
I install nvidia driver by download run file from NVIDIA website which requires not to install opengl files.
`sudo sh ./xxx.run --no-opengl-files`
How can I upgrade my opengl version by this way?
Because when I install nvidia driver by your way, the installation process is always interrupted.

myaldiz, 2023-04-27:
For those who are using laptops, integrated graphics may interfere with correct GL context selection. 
You can run:
```bash
sudo prime-select nvidia
```
to select the Nvidia GPU, and reboot. This fixed my issue.