microsoft/Azure-Kinect-Sensor-SDK #919: Confusion with calculation of dx and dy in transformation_iterative_unproject

foohyfooh, 2019-11-18:
When looking through the code for transformation_iterative_unproject, I noticed something that seemed off in the calculation of the dx and dy on lines [211](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/src/transformation/intrinsic_transformation.c#L211) and [212](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/src/transformation/intrinsic_transformation.c#L212) respectively.

If Jinv is the inverse of the Jacobian matrix J = [df1/dx, df1/dy, df2/dx, df2/fy].
I assumed that the dx would be the result of the functions with respect to x by the error of x rather than both variables and similarly for dy.
So why is the dx = Jinv[0] * err_x + Jinv[1] * err_y instead of dx = Jinv[0] * err_x + Jinv[2] * err_x?
and for dy = Jinv[2] * err_x + Jinv[3] * err_y instead of dy = Jinv[1] * err_y + Jinv[3] * err_y?

Note that I am not that knowledgeable about Jacobians and am just trying to clear up my confusion.

rajeev-msft, 2019-11-19:
**Short explanation:**
The jacobian Jinv, indicates how much a small change in u or v will affect xy. 
x is a function of both u and v and similarly y too. So both terms, du and dv (here it is called err_x and err_y), will determine the change dx. Similarly for dy.

**Longer explanation/ partial derivation:**
*All variables here are same as in the code.

For the project function
p = f(xy) 
where p is the projected point in image coordinates - a 2x1 vector, 
and xy is a 2x1 vector ,
and f is the projection function.

[ px ] = [ fx(xy) ]
[ py ]    [ fy(xy) ]

Where fx and fy are the x and y components of the projection function.

The Jacobian J, for this would be of the form:
J = [ dfx/dx,   dfx/dy ]
      [ dfy/dx,   dfy/dy ]  

[ err_x ] = J * [ dx ]
[ err_y ]         [ dy ]

Hence,
[ dx ] = Jinv * [ err_x ]
[ dy ]              [ err_y ]

where err_x and err_y are  du and dv.

dx = Jinv[0] * err_x  + Jinv[1] * err_y
dy = Jinv[2] * err_x + Jinv[3] * err_y

where Jinv values are in row-major format:
Jinv  = [ Jinv[0],   Jinv[1] ]
           [ Jinv[2],   Jinv[3] ]

*The above derivation lacks some rigor and makes some assumptions, but serves the purpose of this discussion. Refer to Newton-Raphson method for finding the roots of a real-valued function and its usage, for more information on the optimization used.

closed 2019-12-04 by wes-b

shanemankiw, 2020-12-18:
If we are using Brown Conrady model, J is always 0. 
So in this case, we don't need to use the function transformation_iterative_unproject, right?