microsoft/Azure-Kinect-Sensor-SDK #922: Question about Azure Kinect DK recorder recording depth video

1216704928, 2019-11-19:
**Is your feature request related to a problem? Please describe.**

I found that the deep video recorded by the recorder contains IR video. Can the recorder record only the deep video (excluding RGB video and IR video)？

wes-b, 2019-11-19:
Currently Depth and IR are recorded together. But existing MKV tools should allow you to remove tracks.

We would also support adding a `--depth_only` or `--ir_only` flag to the tool so that a recording could be created the way you are asking.

1216704928, 2019-11-20:
Thank you. According to the method you provided, I recorded the depth video containing IR, and then used MKV tool to remove the IR track to get the depth video excluding IR. Can I only see the depth video excluding IR through the K4aviewer provided on the official website?