microsoft/Azure-Kinect-Sensor-SDK #924: Having Trouble Using Green Screen Example

FlyingSpringrol, 2019-11-19:
I've been having some issues aligning the two depth feeds using the boilerplate green screen example:
My process so far:
-Print the OpenCV checkboard from the file provided (which I would not the print settings are not completely specified in the documentation: for example, there is an additional printer setting on mac that is "print entire image" or "fill entire page" that is unspecified in the documentation). I am using print entire image. I think this should be more clear in the docs though.
-Set the program arguments to '2 9 6 22' 
-Press run, and during the calibration phase, move the checkerboard around physically. 
-After the calibrations are calculated, the imshow window opens and things looks very odd. The openCV align function returns an error of ".16777". 

Where I think the documentation could be enhanced to help me and people like me in the future:
-The printing process documentation for the openCV checkboard photo could be a little better (lots more print setting for both windows and mac that could be addressed)
-The whole 'moving the checkerboard' during calibration phase was not clear, I had to read through other people's issues to know that I had to do this. Originally I was just using a static checkerboard. Perhaps adding this to the documentation could be good.
-A way to debug bad calibrations: I noticed the the openCV function returns a "got Error of _____", is there a reference of what is a good error level?

tesych, 2019-11-20:
@FlyingSpringrol, thank you for the feedback. we will add your suggestions to our list of documentation enhancements. As for the good error level, we are looking into it and will share with you as soon as we can.

tesych, 2020-01-23:
documented.

closed 2020-01-23 by tesych

richardeakin, 2020-03-11:
@tesych I'm also searching for answers to a couple of questions the OP posed above around what we should expect in the re-production error, as I don't seem to be getting results very similar to the image included with the example (I'd be happy to share these). Questions:

- What is the ideal physical camera configuration for running in multi-device mode? Also, what is the configuration that the screenshot was taken with
- What is an acceptable and / or expected re-production error that we are seeing in the return from `cv::stereoCalibrate(()`?

Thanks!

qbitronics, 2020-12-20:
@tesych Would be really helpful if you could guide me through this.
i) I have connected 3 azure kinect  devices and synced them in daisy chain structure through 3.5 mm audio cables.
Question is :
a)How could i make them work together and get 3D point cloud data in three dimensions?
b) How to make Master communicate with other two Subordinates and get data in 3 dimensions.

ii) Ran the green screen example program.
Question is :
a)How to fix positions for chessboard for three azure kinect sensors and calibrate the camera?
Would be of use if you could look into this. and could just hint me anything.

 Thanks in advance.