microsoft/Azure-Kinect-Sensor-SDK #925: Calibration

TxT1212, 2019-11-20:
**Describe the bug**

**To Reproduce**

**Expected behavior**

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: 
 - SDK Version:  
 - Firmware version:  

**Additional context**

TxT1212, 2019-11-20:
BTW, what is the use of k4 k5 k6? It seems OpenCV calibration result only has k1 k2 k3. 
I found some related code in k4a.h 
`K4A_EXPORT k4a_result_t k4a_calibration_color_2d_to_depth_2d(const k4a_calibration_t *calibration,
                                                             const k4a_float2_t *source_point2d,
                                                             const k4a_image_t depth_image,
                                                             k4a_float2_t *target_point2d,
                                                             int *valid);`
but I couldn't find how they are used.

amonnphillip, 2019-11-20:
@TxT1212  Opencv 3+ supports all 8 distortion parameters used by the Azure Kinect dk camera.

What version of Opencv are you using?

TxT1212, 2019-11-21:
> @TxT1212 Opencv 3+ supports all 8 distortion parameters used by the Azure Kinect dk camera.
> 
> What version of Opencv are you using?

well, I asked my computer  `pkg-config --modversion opencv
`
and my computer told me I am using 3.2.0. Because I installed multi versions of opencv, so I'm not sure which version ros used. ros version is malena.

jasjuang, 2019-11-26:
Not familiar with ros but you can test it out simply with just opencv. You have to use all 8 distortion parameters and put them in the order of k1 k2 p1 p2 k3 k4 k5 k6 and then pass it to `cv::undistort`.

TxT1212, 2019-12-05:
> Not familiar with ros but you can test it out simply with just opencv. You have to use all 8 distortion parameters and put them in the order of k1 k2 p1 p2 k3 k4 k5 k6 and then pass it to `cv::undistort`.

thank you for your reply, I found the function. I calibrate the kinect myself, the result seems much better.

closed 2019-12-05 by TxT1212