microsoft/Azure-Kinect-Sensor-SDK #926: DecodeMJPEGtoBGRA32(). MJPEG decode failed: -1?

nonlinear1, 2019-11-20:
When I run two kinect for  azure in a computer at the sametime with K4A_WIRED_SYNC_MODE_STANDALONE mode;
sometimes, the terminal give me the error:
[2019-11-20 12:46:12.284] [error] [t=8802] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/color/uvc_camerareader.cpp (1300): DecodeMJPEGtoBGRA32(). MJPEG decode failed: -1
[2019-11-20 12:46:12.284] [error] [t=8803] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (269): capturesync_get_capture(device->capturesync, capture_handle, timeout_in_ms) returned failure in k4a_device_get_capture()
I try to avoid this error, but I failed.
the enviroment is as follow:
my os ubuntu18.04
gpu: gtx1070
ram:16G
cpu: i7
and the driver and firmware of my kinect is the newest one!
anyone know how to solve this problme? Thank you very much!

wes-b, 2019-11-20:
Probably unrelated, but have you increased your USB memory?

nonlinear1, 2019-11-20:
I have increased usb memory according to "Multiple Device Sync Issues". I think maybe this is caused by hardware, I change another usb port. It is no this error!

wes-b, 2019-11-20:
I see your last comment was:
> I have increased usb memory according to "Multiple Device Sync Issues". I think maybe this is caused by hardware, I change another usb port. It is no this error!

and now it is:
>I have increased usb memory according to "Multiple Device Sync Issues", the same error exist!.

Can you share what changed? is this a USB port issue? There are known issues with USB Host Controllers, so it is very possible one USB port works and another does not.

wes-b, 2019-11-20:
The actual error is comming from tjDecompress2() which belongs to libjpeg-turbo. The error code -1 but not helpful, so we will need to dig into that code.

Is this issue happen 100% of the time? Does it happen immediately after starting the camera?

nonlinear1, 2019-11-20:
Actually, yesterday I change the kinect which has error message,  to another USB port and the error is disappeared.  I want to test and verify this is related to USB port. the issue  happens sometimes not 100% of time. It happen after the camera running for a while not after starting the camera.

wes-b, 2019-11-21:
Curious about your cabling and power setup. Are you using all provided cables? USB extender? Correct power supply?

nonlinear1, 2019-11-22:
I am testing and try to solve this problem, wait for me, one day, I will give you more information! Thank you for your help!

nonlinear1, 2019-11-22:
OK, I give my cable and power information:
first, I use the cable including the power and data , which is provided by microsoft and then I use two usb3.0 extension cable about five metres to connect the cables, which is provided by mircosoft. Last, the power usb3.0 extension cable is connected the adapter provided by microsoft, and the data usb3.0 extension cable is connected to usb3.0 port of my computer. Both kinects is the same connection!

wes-b, 2019-11-22:
Even though you are using multiple USB extension cables, I do not think that is the problem or you would be seeing USB errors. 

I suspect you are getting some kind of corruption of the image. We have not seen this issue reported from anywhere else, so I would look to your code for memory leaks or out of bounds memory access.

If that doesn't turn anything up then you can modify the function save the image to disk, so that we can analyze it.

nonlinear1, 2019-11-22:
OK,I will try

wes-b, 2019-12-04:
@nonlinear any update on this issue? Can we close this?

nonlinear1, 2019-12-05:
OK, you can close. recently, I have not received the error. Thank you for your help!

closed 2019-12-05 by wes-b