microsoft/Azure-Kinect-Sensor-SDK #927: Can the recorder be changed to record MKV video (including audio)？

1216704928, 2019-11-20:
**Is your feature request related to a problem? Please describe.**

Can the recorder be changed to record MKV video (including audio)？
**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

tesych, 2019-11-20:
@1216704928 , k4arecorder doesn't record sound. You can use Speech SDK to do so. You can find more documentation as part of our [Azure Kinect documentation](https://docs.microsoft.com/en-us/azure/Kinect-dk/access-mics). Currently there's no plan to include the audio. If this feature is important for you, please add this to the [Customer voice](https://feedback.azure.com/forums/920053) so it can be upvoted.

closed 2019-11-20 by tesych

1216704928, 2019-11-21:
Can I capture the audio from the k4aviewer?

wes-b, 2019-12-20:
> Can I capture the audio from the k4aviewer?

Sorry, no you can't

TristanWYL, 2021-09-08:
Check [this](https://github.com/TristanWYL/Azure-Kinect-Recorder) out, I believe you could find what you need.