microsoft/Azure-Kinect-Sensor-SDK #930: How is the IR image in passive mode (1024x1024) turned into NFOV_UNBINNED mode (640x576)

jasjuang, 2019-11-21:
I am trying to calibrate the IR camera. I captured the checkerboard images for the IR cameras in passive mode (K4A_DEPTH_MODE_PASSIVE_IR), and it has a resolution of 1024x1024, but I captured the actual subject in NFOV_UNBINNED mode, which has a resolution of 640x576. Can someone let me know what is the relationship between the images at two modes? Is it some sort of crop/scale/offset?

I tried cropping 192 pixels on the width for both sides and 224 pixels on the height for both sides on the 1024x1024 image but it's different from the 640x576 image from NFOV_UNBINNED mode.

Related to #921 and #803.

rabbitdaxi, 2019-12-02:
Sounds like you want to recalibrate your device, you can refer to the following SDK code to understand how the crop offset was applied inside the SDK mode specific intrinsics computation. https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/90f77529f5ad19efd1e8f155a240c329e3bcbbdf/src/transformation/mode_specific_calibration.c#L95

jasjuang, 2019-12-02:
I manually shift the image pixel by pixel and wasn't sure whether the correct offset is 180 or 179 in the vertical direction. Thanks for confirming it is 180.

closed 2019-12-02 by jasjuang

xiaodongww, 2020-11-11:
Sorry to reopen this issue. I also meet some problems in recalibrating. I am new to the RGB-D camera and calibration process. It would be very nice if anyone could give me some suggestions.

I am trying to recalibrate the color and IR camera (two camera's intrinsic, distortion parameter and two cameras' relative rotation and translation). For the intrinsic and distortion, I can calculate it successfully using `OpenCV chessboard calibration`. There are two questions about rotation and translation:
- Do I need to **undistort the chessboard image**(both Color and IR chessboard image) before calculating the extrinsic(i.e. rotation and translation between Color and IR camera)?
- How to calculate the `extrinsic`? Is there any tool to use?

Any suggestions will be helpful, thanks very much. 
BTW, please let me know if I was supposed to open a new issue.