microsoft/Azure-Kinect-Sensor-SDK #931: kinect azure rgb flow

barakooda, 2019-11-21:
can you please elaborate on the kinect azure rgb flow ?

I guess its goes this way :
after debayer and ISP --> RGB ( native ) -> FPGA ( YUV\MJPEG) or leave it as RGB > thru USB 3.0 -> system.

but its written the sdk use by default the mjpeg option.
Do i understand right ?

wes-b, 2019-11-21:
Image format is captured natively in YUY2, NV12, or MJPG by the sensor. It is transferred to the PC via USB3.0. On Windows Media Foundation is used to ready the UVC end point, on Linux LibUvc is used.

barakooda, 2019-11-21:
When you say "sensor" you mean the whole kinect ?
cause the RGB sensor in the kinect capture rgb 10bit "regular" (bayered) rgb matrix.
it must be hardware that convert the RGB (after debayer) to YUV and Mjpeg....

what i`m missing ?

wes-b, 2019-11-21:
We have not documented these details. If you are blocked then please let us know what is blocking you.

barakooda, 2019-11-21:
I`m just confused.
the kinect azure viewer enable you 4 color modes with different resolutions#fps etc...
in the hardware spec docs it shows very strict table with MJPEG ruling most of it.

if the kinect viewer is just sdk ui interface - why the table does not reflect this options too ?

anyway, 
I`m just make sure I can get RGB, MJPEG and NV12 color modes for my use.

wes-b, 2019-11-21:
The K4aViewer is showing the same combination of resolution, format, and frame rates that are documented here: https://docs.microsoft.com/en-us/azure/Kinect-dk/hardware-specification#color-camera-supported-operating-modes. These are the native resolution and formats supported by the rgb sensor.

Additionally the SDK (and viewer) add a BGRA formats, which the SDK does the conversion.

barakooda, 2019-11-22:
Thank you very much for your quick replies .

closed 2019-11-22 by wes-b