microsoft/Azure-Kinect-Sensor-SDK #933: a strange and confusing phenomenon with undistorting the depth and color image

nonlinear1, 2019-11-22:
I add undistortion part of color image and depth image to get undistort color image and depth image.  And, I use the undistort color image and depth to make a pointcloud file, which is pcd format. But the pointcloud file has many scattering points along the ray, which is emitted by depth camera. I am confusing this  phenomenon.
the pointcloud srceen shot is as follow:
[image]
and the following is my codes:
#include <iostream>
#include <k4a/k4a.h>
#include <opencv2/core/core.hpp>
#include <opencv2/highgui/highgui.hpp>
#include <opencv2/imgproc/imgproc.hpp>
#include <iomanip>
#include <opencv2/calib3d/calib3d.hpp>
using namespace std;
using namespace cv;

template<typename T> Mat create_mat_from_buffer(T *data, int width, int height, int channels = 1)
{
    Mat mat(height, width, CV_MAKETYPE(DataType<T>::type, channels));
    memcpy(mat.data, data, width * height * channels * sizeof(T));
    return mat;
}

static string get_serial(k4a_device_t device)
{
    size_t serial_number_length = 0;

    if (K4A_BUFFER_RESULT_TOO_SMALL != k4a_device_get_serialnum(device, NULL, &serial_number_length))
    {
        cout << "Failed to get serial number length" << endl;
        k4a_device_close(device);
        exit(-1);
    }

    char *serial_number = new (std::nothrow) char[serial_number_length];
    if (serial_number == NULL)
    {
        cout << "Failed to allocate memory for serial number (" << serial_number_length << " bytes)" << endl;
        k4a_device_close(device);
        exit(-1);
    }

    if (K4A_BUFFER_RESULT_SUCCEEDED != k4a_device_get_serialnum(device, serial_number, &serial_number_length))
    {
        cout << "Failed to get serial number" << endl;
        delete[] serial_number;
        serial_number = NULL;
        k4a_device_close(device);
        exit(-1);
    }

    string s(serial_number);
    delete[] serial_number;
    serial_number = NULL;
    return s;
}

int main(int argc, char **argv)
{ 

    cout << endl << "-------" << endl;
    cout << "Openning Kinect for Azure ..." << endl;
    cv::Mat imBGRA, imBGR, imD, im_transformed_color_image;
    cv::Mat im_transformed_depth_image;
   

    cv::Mat camera_matrix;
    cv::Mat new_camera_matrix;
    float factor{ 1.0 }; // scaling factor
    cv::Mat cv_undistorted_color;
    cv::Mat cv_undistorted_depth;
    cv::Mat cv_depth_downscaled;
[... 249 lines cut ...]
	  cv_color_downscaled.size(),
	  INTER_LINEAR);//CV_INTER_AREA

          remap(cv_depth_downscaled, cv_undistorted_depth, map1, map2, cv::INTER_LINEAR, cv::BORDER_CONSTANT);
	  remap(cv_color_downscaled, cv_undistorted_color, map1, map2, cv::INTER_LINEAR, cv::BORDER_CONSTANT);
          cv::cvtColor(cv_undistorted_color, cv_undistorted_color, CV_BGRA2RGB);
	

           imwrite(file, cv_undistorted_depth, compression_params);
           imwrite(file1, cv_undistorted_color, compression_params);

           k4a_image_release(depth_image); 
           k4a_image_release(color_image);

           k4a_image_release(transformed_depth_image); 
           k4a_capture_release(capture);
           ++count;
    
    }

     return 0;
    
}

Could anyone know why? Thanks a lot

jasjuang, 2019-11-23:
In your `remap` you are using `INTER_LINEAR`, which will cause weird artifacts at depth discontinuity. You have to use `INTER_NEAREST` when you warp or interpolated depth, or if you care about the depth quality you can do some sort of depth discontinuity detection where at non depth discontinuity use `INTER_LINEAR` and depth discontinuity use `INTER_LINEAR`.

nonlinear1, 2019-11-23:
I will try thank you very much

nonlinear1, 2019-11-23:
@jasjuang Thank you for your help, you are right!

closed 2019-11-23 by wes-b

SeanStephensenAIRM, 2020-03-11:
@nonlinear1 @jasjuang Commenting here because my issue seems to be identical. Please let me know if this is bad form and I should start a fresh issue instead. I'm post processing k4a images in python and running into the same error. I'm reading into python:

- 24 bit, 3 channel **color** image, uint8 dtype

- 16 bit 1 channel **depth** image, uint16 dtype

- the Kinect camera's intrinsic and distortion data

The images are reading in correctly, and generating a point cloud from the raw images works perfectly fine. They I apply the following code:

```
h, w = color_im.shape[:2]
    newcameramtx, roi = cv2.getOptimalNewCameraMatrix(matrix_in, dist_vector, (w, h), 1, (w, h))
    x, y, w, h = roi
    mapx, mapy = cv2.initUndistortRectifyMap(matrix_in, dist_vector, None, newcameramtx, (w, h), 5)
    depth_undistorted = cv2.remap(depth_im, mapx, mapy, interpolation=cv2.INTER_NEAREST)
    color_undistorted = cv2.remap(color_im, mapx, mapy, interpolation=cv2.INTER_LINEAR)
    color_cropped = color_undistorted[y:y+h, x:x+w]
    depth_cropped = depth_undistorted[y:y+h, x:x+w]

    color_converted = o3d.geometry.Image(color_cropped)
    depth_converted = o3d.geometry.Image(depth_cropped)
    combined_im = o3d.geometry.RGBDImage.create_from_color_and_depth(color_converted, depth_converted, depth_scale=1000.0, depth_trunc=1.0, convert_rgb_to_intensity=False)
    combined_pcd = o3d.geometry.PointCloud.create_from_rgbd_image(combined_im, matrix_in)
```

The color image comes out fine, and the depth image is hard to check since it's 16 bit. Here is the raw color image (left) and the undistorted color image (right)

[image]

Now, here's a comparison of the "raw" point cloud vs the undistorted point cloud. Obviously something gone wrong, and I get the same "pyramidal" shape to the point cloud that @nonlinear1 seemed to encounter:
[image]
[image]

As you can see, I'm already specifying NEAREST interpolation for depth undistortion, and still running into the problem. Any other ideas?