microsoft/Azure-Kinect-Sensor-SDK #935: Why is the frame rate so low

CaoZheng123, 2019-11-22:
I used the bodyskd tool to run smoothly, but when I used the code to run it the FPS was low.

The former is close to 30fps, while the latter is about 15.

tesych, 2019-11-22:
@CaoZheng123 can you please provide more information about your scenario, platform and what exactly to help us answer your question.

CaoZheng123, 2019-11-23:
> @CaoZheng123 can you please provide more information about your scenario, platform and what exactly to help us answer your question.

> @CaoZheng123 can you please provide more information about your scenario, platform and what exactly to help us answer your question.

ok。I'm using this code right here：{Azure-Kinect-Samples/body-tracking-samples/simple_3d_viewer/}。
I also made sure that CPU only was false，and the interface that this code runs on is poor。

wes-b, 2019-12-02:
@CaoZheng123 please look at our **_bug_** template by clicking on **_New issue_** and then **_Sensor SDK Bug report_**. To help we need to know OS, repro steps, how you are building, and other key info. My guess is that you are building debug instead of retail and that has a performance hit.

CaoZheng123, 2019-12-04:
@wes-b 
Yes, your guess is correct.When I used Release instead of Debug, performance was fine

closed 2019-12-04 by wes-b