microsoft/Azure-Kinect-Sensor-SDK #938: Ubuntu repo's are returning some 404's.

R0NAM1, 2019-11-24:
(Following the official instructions: https://docs.microsoft.com/en-us/azure/kinect-dk/sensor-sdk-download) I am getting a 404 error when running apt update, and turns out it's looking for https://packages.microsoft.com/ubuntu/19.04/prod/dists/disco/main/binary-i386/Packages which doesn't exist as the structure end at /main, I know nobody runs i386 with Ubuntu (That's just mad) but I am not able to apt install k4a-tools,

Ubuntu 19.04
SDK Version - NA
Firmware Version - NA

RoseFlunder, 2019-11-24:
As far as I know Ubuntu 18.04. is the only supported distribution for the Azure Kinect SDK.

R0NAM1, 2019-11-26:
> 
> 
> As far as I know Ubuntu 18.04. is the only supported distribution for the Azure Kinect SDK.

When I try to sync those packages with my system, I get the same error, https://packages.microsoft.com/ubuntu/18.04/prod/dists/disco/main/binary-i386/Packages is still a 404.

RoseFlunder, 2019-11-26:
I don't know what this URL should do.
I followed this instructions for Ubuntu 18.04. and it works.
Ubuntu Disco is not supported. There are no Ubuntu binaries other than that for Ubuntu 18.04.

R0NAM1, 2019-12-01:
I was able to solve by manually downloading the .deb files and installing them from the repo.

closed 2019-12-01 by R0NAM1