microsoft/Azure-Kinect-Sensor-SDK #940: Current setting value can't get when automatic control mode

UnaNancyOwen, 2019-11-25:
**Describe the bug**

The current setting value of exposure-time and whitebalance can't get in automatic control mode.
They are automatically adjusted in automatic control mode, but the current values ​​that get with <code>k4a_device_get_color_control()</code> don't change.

**To Reproduce**

We confirmed with the following procedure.
1. Set Manual Mode and 500μs to Exposure Time for initialize.
    If you get the value here, it is 500μs. Of course, the color image is dark.
2. Set Auto Mode to Expousure Time.
    The color image is automatically adjusted and brightened.
    The value should have changed to big value than first internally.
3.  If you get a value again here, it is 500μs yet. Why?

**Expected behavior**

It should return the current value even in automatic mode.

**Logs**

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS with Version: Windows 10 (1903)
 - SDK Version:  1.3.0
 - Firmware version:  1.6.102 (RGB Camera Firmware)

**Additional context**

UnaNancyOwen, 2019-11-26:
This is minimum code to reproduce this problem.
```cpp
#include <iostream>
#include <chrono>
#include <conio.h>
#include <k4a/k4a.hpp>

std::string to_string( k4a_color_control_mode_t mode )
{
    return ( ( mode == K4A_COLOR_CONTROL_MODE_AUTO ) ? "Auto" : "Manual" );
}

int main( int argc, char* argv[] )
{
    k4a::device device = k4a::device::open( K4A_DEVICE_DEFAULT );

    k4a_device_configuration_t device_configuration = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
    device_configuration.color_format     = K4A_IMAGE_FORMAT_COLOR_BGRA32;
    device_configuration.color_resolution = K4A_COLOR_RESOLUTION_720P;
    device.start_cameras( &device_configuration );

    // Set Control Mode to Manual Mode and Exposure Time to 500 microseconds
    std::chrono::microseconds exposure_time( 500 );
    device.set_color_control( K4A_COLOR_CONTROL_EXPOSURE_TIME_ABSOLUTE, K4A_COLOR_CONTROL_MODE_MANUAL, exposure_time.count() );

    // Get Current Control Mode and Exposure Time
    int32_t value;
    k4a_color_control_mode_t mode;
    device.get_color_control( K4A_COLOR_CONTROL_EXPOSURE_TIME_ABSOLUTE, &mode, &value );

    /* "Manual 500" is displayed on console */
    std::cout << to_string( mode ) << " " << value << "\n";

    // Set Control Mode to Auto Mode
    device.set_color_control( K4A_COLOR_CONTROL_EXPOSURE_TIME_ABSOLUTE, K4A_COLOR_CONTROL_MODE_AUTO, exposure_time.count() );

    while( !_kbhit() ){
        k4a::capture capture;
        device.get_capture( &capture );
        capture.reset();

        // Get Current Control Mode and Exposure Time
        int32_t value;
        k4a_color_control_mode_t mode;
        device.get_color_control( K4A_COLOR_CONTROL_EXPOSURE_TIME_ABSOLUTE, &mode, &value );

        /* "Auto 500" is displayed on console all the time */
        std::cout << to_string( mode ) << " " << value << "\n";
    }

    device.stop_cameras();
    device.close();

    return 0;
}
```

wes-b, 2020-07-10:
This should be fixed with the newest firmware; AzureKinectDK_Fw_1.6.110079014.bin.

closed 2020-07-11 by qm13