microsoft/Azure-Kinect-Sensor-SDK #941: SDK dropping depth frames

t-walker-21, 2019-11-25:
**Describe the bug**

When using the azure, I am constantly getting warnings from both ROS and the k4aviewer. The warning arises from the depth_engine_thread, and it states that the processing of the depth image is too slow @ 1000ms. I have encountered this issue on multiple machines even when running the camera at 5 fps. The sdk recommends a firmware update, how can I acquire the binary that needs to be passed into the firmware tool?

**To Reproduce**

wes-b, 2019-12-05:
There are a couple issues here:
- There is a bug in the reporting of 1000ms that we need to fix.
- You ask about the binary needed for the firmware tool.
  - The binary is in the [Installer](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md#msis) and in the [NuGet](https://www.nuget.org/packages/Microsoft.Azure.Kinect.Sensor/1.3.0) packages.
  - If you have more questions about this, then lets open a new issue.
- dropped frames are typically related to PC performance. Perhaps you can enable logging at an informational level and we can help confirm that version another issue.

closed 2019-12-18 by wes-b

wes-b, 2020-01-17:
https://github.com/Azure/azure-c-shared-utility/pull/423