microsoft/Azure-Kinect-Sensor-SDK #942: The [Azure- kinect-samples /simple_3d_viewer] file was compiled in vs2019 with a low frame count

CaoZheng123, 2019-11-26:
I used the exe file of Azure Kinect Body Tracking SDK/tool at the beginning and it runs very smoothly.
But if you compile with the code for Azure- kinect-samples/body-track-samples /, it's very slow and slow。
code：
https://github.com/CaoZheng123/Azure-Kinect-Samples/tree/master/body-tracking-samples/simple_3d_viewer.
I started with a gtx1070 graphics card，I suspect it may be the performance of the graphics card.
But today I ran it on an rtx2080 graphics card and it got better, but the delay was still very high.

Both computers run the programs in the SDK /tool smoothly
I would like to ask Azure Kinect Body Tracking SDK/tool program and the code compiled program is not the same?

Chris45215, 2019-11-30:
I had the same problem and thought the exact same thing when I first tried it, then found this solution. If you are using Visual Studio, make sure that the Solutions Configurations dropdown selection (it is to the left of the "Run" button and the "Any CPU" dropdown) is set for "Release" rather than "Debug".
That should solve it.

CaoZheng123, 2019-12-01:
@Chris45215    
Thank you very much. You solved my problem

closed 2019-12-02 by wes-b