microsoft/Azure-Kinect-Sensor-SDK #943: Sync Out channel pulses don't correspond to depth frames

neurojaym, 2019-11-26:
**Describe the bug**
<I'm sending a cable from the sync out channel to an Arduino/Teensy in order to control some external electronics and timelock signals between the Azure and another device. However, I am having a consistent problem, when I run a 10min/600sec session and designate the k4 as the master for an external sync, I find that the saved .mkv file contains 17970 depth frames (at 30Hz) instead of 18000. Additionally, my Arduino set up detects 18018 pulses emerging from the sync out channel, suggesting perhaps that the sync out channel continues sending pulses after frames have finished collection and have been saved and right up until the camera has been stopped. I'm wondering if developers have any clear measurements of why this mismatch may be occurring--namely 1) does the k4 stop collecting frames before the designated time is up, 2) are there pulses sent out after the final frames have been collected that may account for the mismatch?>

**To Reproduce**

**Expected behavior**
<I expected that for a 600s session, I would collect precisely 18000 frames and that an Arduino would detect exactly 18000 sync out pulses.>

**Logs**

**Screenshots**
<*Ideal* Set Recording Frames 18000.0
Teensy Recorded Pulses: 18018
K4 Acquired Frames: 17970
Actual Time sampled by Teensy: 600.6
Actual Time sampled by K4: 599.0>

**Desktop (please complete the following information):**
 - OS with Version: <Windows 10 Pro>
 - SDK Version:  < v1.3.0>
 - Firmware version:  <RBG=1.6.102, Depth=1.6.75 >

**Additional context**

wes-b, 2019-11-27:
The device doesn't produce a perfect stream of captures immediate after start. It needs a second or so to stabilize. I would expect 1s to 601s to give you the expected results

jmarkow, 2019-12-02:
@wes-b I see.  So is there a recommended procedure for synchronizing other pieces of equipment to the Kinect? If the pulses from sync out matched the frames 1:1 this would be fairly trivial.

jmarkow, 2019-12-02:
e.g. if the camera were to drop a bunch of frames during the first second of the recording, is there a way to know which frame in the mkv file corresponds to the 500th pulse?

wes-b, 2019-12-02:
> So is there a recommended procedure for synchronizing other pieces of equipment to the Kinect? If the pulses from sync out matched the frames 1:1 this would be fairly trivial.

We are working on better documentation for this. The pulse is the V-Sync for color. And if the depth_delay_off_color is 0, then it it effectively the start of the depth capture too.

> e.g. if the camera were to drop a bunch of frames during the first second of the recording, is there a way to know which frame in the mkv file corresponds to the 500th pulse?

No, the dropped frames on start are not 100% predictable.

jmarkow, 2019-12-03:
I see.  In the mkv files saved by k4arecorder is it possible to determine where frames were dropped?  That is, do the timestamps correspond to the device timestamps with the onset time removed? If that's the case, then we could potentially use sync out for alignment to external equipment if I understand things correctly.  That is, if there are 100 pulses from sync out and 2 frames were dropped (and we know where they were dropped exactly), then pulse 100 should be aligned with frame # 98 in the mkv file?

wes-b, 2019-12-03:
The SDK provides both a device timestamp and a system timestamp. The device timestamp can be used to determine where a frame might have been dropped. So yes if you know you dropped 2, (and yes you will know where they are at) you should be able to correspond pulse 100 to image #98 in the mkfile.

jmarkow, 2019-12-03:
Great! Hopefully I didn't miss this–is there documentation on the format used to store the system and device timestamps in the mkfile? We're happy to use the SDK functions, but curious if there are simple ways to extract the timestamps using command line tools (e.g. ffmpeg or mkvtoolnix).

xthexder, 2019-12-03:
The device timestamp can be calculated by adding the `K4A_START_OFFSET_NS` tag to the raw frame timestamp in the recording.
Both ffmpeg and mkvinfo will show the recording tags. mkvinfo will output individual timestamps and a bunch of other info if you pass `-v`, and I believe you can get timestamps out of ffmpeg as well.

If you end up using the playback api, all timestamps returned will be in device time already.

hagorms, 2020-05-22:
> The SDK provides both a device timestamp and a system timestamp. The device timestamp can be used to determine where a frame might have been dropped. So yes if you know you dropped 2, (and yes you will know where they are at) you should be able to correspond pulse 100 to image 98 in the mkfile.

Do you have an example code to do it?

LuyaooChen, 2021-07-12:
Hi, I am tring to record two or more .mkv files with external sync. I set the same recording time param to `k4arecorder`, but the mkv files have different `Recording Length`(shown in k4aviewer). And I extract images using ffmpeg, the number of images from different mkv file are different. 
How to confirm which frame is dropped?

closed 2022-05-14 by qm13