microsoft/Azure-Kinect-Sensor-SDK #945: RGB camera error when connect the azure kinect through usb-c cable only

lesterlo, 2019-11-29:
**Describe the bug**
The k4a viewer receive errors (camera time out error)the azure kinect device when only use a usb-c cable to connect to the computer. The kinect didn't have any problem when visualize depth data only. The problem will occur when trying to visualize RGB data.

**To Reproduce**
1. Connect the usb-c cable
2. Azure kinect no amber light, only white light
3. Open k4a viewer
4. the k4a viewer  can found the device with serial number displayed
4. when press open device -> (no problem)
5. press start button (-> problem occured) 

**Expected behavior**
I can receive all kinect frame including depthm, rgb, etc.

**Logs**

```bash
# with depth + rgb camere

[ warning  ] : is_fw_version_compatable(). Firmware version for RGB is 1.6.98. Consider upgrading to 1.6.102 or newer.
[ warning  ] : is_fw_version_compatable(). Firmware version for Depth is 1.6.70. Consider upgrading to 1.6.75 or newer.
[ trace    ] : k4a_device_start_cameras(). k4a_device_start_cameras starting
[ info     ] : k4a_device_start_cameras(). Starting camera's with the following config.
[ info     ] : k4a_device_start_cameras().     color_format:3
[ info     ] : k4a_device_start_cameras().     color_resolution:1
[ info     ] : k4a_device_start_cameras().     depth_mode:2
[ info     ] : k4a_device_start_cameras().     camera_fps:2
[ info     ] : k4a_device_start_cameras().     synchronized_images_only:1
[ info     ] : k4a_device_start_cameras().     depth_delay_off_color_usec:0
[ info     ] : k4a_device_start_cameras().     wired_sync_mode:0
[ info     ] : k4a_device_start_cameras().     subordinate_delay_off_master_usec:0
[ info     ] : k4a_device_start_cameras().     disable_streaming_indicator:0
[ trace    ] : usb_cmd_io(). XFR: Cmd=80000001, CmdLength=13, PayloadSize=0, CmdData=00000000 00000000...
[ trace    ] : usb_cmd_io(). XFR: Cmd=000000e1, CmdLength=4, PayloadSize=0, CmdData=00000004 6829a550...
[ trace    ] : usb_cmd_io(). XFR: Cmd=00000022, CmdLength=4, PayloadSize=2000000, CmdData=00000002 6829a550...
[ trace    ] : dynlib_t_create(). Created   dynlib_t 0x7fc244000b50
[ info     ] : verify_plugin(). Loaded K4A Plugin with version: 2.0.0
[ trace    ] : usb_cmd_io(). XFR: Cmd=00000103, CmdLength=4, PayloadSize=0, CmdData=0000001e 6829a550...
[ trace    ] : usb_cmd_io(). XFR: Cmd=00000009, PayloadSize=0
[ trace    ] : usb_cmd_io(). XFR: Cmd=000000f1, PayloadSize=0
[ trace    ] : k4a_image_t_create(). Created   k4a_image_t 0x7fc238000c90
[ trace    ] : k4a_capture_t_create(). Created   k4a_capture_t 0x7fc238000e90
[ trace    ] : k4a_image_t_create(). Created   k4a_image_t 0x7fc238001020
[ trace    ] : k4a_capture_t_create(). Created   k4a_capture_t 0x7fc238001220
[ trace    ] : k4a_image_t_create(). Created   k4a_image_t 0x7fc2380013b0
[ trace    ] : k4a_capture_t_create(). Created   k4a_capture_t 0x7fc2380015b0
[ trace    ] : k4a_image_t_create(). Created   k4a_image_t 0x7fc238001740
[ trace    ] : k4a_capture_t_create(). Created   k4a_capture_t 0x7fc238001940
[ trace    ] : k4a_image_t_destroy(). Destroyed k4a_image_t 0x7fc238001020
[ trace    ] : k4a_capture_t_destroy(). Destroyed k4a_capture_t 0x7fc238001220
[ trace    ] : k4a_image_t_create(). Created   k4a_image_t 0x7fc238001d00
[ trace    ] : k4a_capture_t_create(). Created   k4a_capture_t 0x7fc244c95490
[ trace    ] : k4a_image_t_create(). Created   k4a_image_t 0x7fc245704eb0
[ trace    ] : k4a_image_t_create(). Created   k4a_image_t 0x7fc245705040
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS: 149184388
[ trace    ] : k4a_image_t_destroy(). Destroyed k4a_image_t 0x7fc245704eb0
[ trace    ] : k4a_image_t_destroy(). Destroyed k4a_image_t 0x7fc245705040
[ trace    ] : k4a_capture_t_destroy(). Destroyed k4a_capture_t 0x7fc244c95490
[ trace    ] : k4a_capture_t_destroy(). Destroyed k4a_capture_t 0x7fc238000e90
[ trace    ] : k4a_image_t_destroy(). Destroyed k4a_image_t 0x7fc238000c90
[ warning  ] : queue_pop(). dewrapper: Dropped oldest 1 captures from queue
[ trace    ] : k4a_capture_t_create(). Created   k4a_capture_t 0x7fc238512a90
[ trace    ] : k4a_image_t_create(). Created   k4a_image_t 0x7fc238000c90
[ trace    ] : k4a_capture_t_create(). Created   k4a_capture_t 0x7fc244c2e3c0
[ trace    ] : k4a_image_t_create(). Created   k4a_image_t 0x7fc244c2e550
[ trace    ] : k4a_image_t_create(). Created   k4a_image_t 0x7fc244c2e6e0
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS: 149251066
[ trace    ] : k4a_image_t_destroy(). Destroyed k4a_image_t 0x7fc244c2e550
[ trace    ] : k4a_image_t_destroy(). Destroyed k4a_image_t 0x7fc244c2e6e0
[ trace    ] : k4a_capture_t_destroy(). Destroyed k4a_capture_t 0x7fc244c2e3c0
[... 647 lines cut ...]
[ warning  ] : queue_disable(). Waiting for blocking call to complete
[ warning  ] : queue_error(). Error detected, shutting down queue and notifying consumers
[ warning  ] : queue_error(). Error detected, shutting down queue and notifying consumers
[ warning  ] : queue_error(). Error detected, shutting down queue and notifying consumers
[ warning  ] : capturesync_add_capture(). Capture Error Detected, Depth 
[ info     ] : k4a_device_stop_cameras(). k4a_device_stop_cameras stopped
```

**Screenshots**

* Error message*
[image]

*Depth Only*
[image]

*RGB only*
[image]

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 18.04.3
 - SDK Version:  1.2.0
 - Firmware version:   Firmware version for RGB is 1.6.98. Firmware version for Depth is 1.6.70. 

**Additional context**

wes-b, 2019-12-02:
I don't think this is a USB C issue. If you plug in the extra power supply can you still reproduce the issue?

I don't see an error in the log that would explain why you aren't getting color images. Are you seeing debug output going to the console window you launched K4aViewer from? If so that is LibUvc debug output and that would be useful to see as well.

lesterlo, 2019-12-03:
Yes, when I plug in the usbc cable with the extra power supply, I can still reproduce the issue.

The strange behavior is that I can use the usbc cable to start all of the sensor including depth camera, IMU and microphone except **RGB camera** .

I am using the K4aViewer's log to output the log message. I set the Severity level to Info. 

What shout I do to output another useful debug message?

lesterlo, 2019-12-03:
I solve the problem. 

When I plug the azure kinect to the USB-c port at the computer's back panel. Everything is normal. I can obtain all of the sensor data without any problem. 

Maybe the USB port on the front panel is not compatible with Kinect. Btw, I am using Zotac En72070V.

closed 2019-12-03 by lesterlo

wes-b, 2019-12-03:
> The front panel's usb ports on Zotac En72070V is using ASMedia Technology Inc chip.
> only the back panel usb ports is using Intel usb3.1 chip

Great info, thanks for sharing!