microsoft/Azure-Kinect-Sensor-SDK #946: Kinect camera fps cannot reach 30fps

Linstudio36, 2019-11-30:
**Describe the bug**
I used the code in https://docs.microsoft.com/en-us/azure/kinect-dk/build-first-body-app and I also add deviceConfig.camera_fps = K4A_FRAMES_PER_SECOND_30; in the camera device setup. 
However, when I print out the time stamp of each body frame (k4abt_frame_get_device_timestamp_usec(body_frame)) , I see the time difference between each timestamp is 66666 instead of 33333. If the timestamp difference is 66666, it means the Kinect is functioning with 15fps instead of 30fps.

**To Reproduce**

1. Go to https://docs.microsoft.com/en-us/azure/kinect-dk/build-first-body-app, copy the Full source code at the bottom of the page to Visual Studio

2.  Add `deviceConfig.camera_fps = K4A_FRAMES_PER_SECOND_30;` under `deviceConfig.color_resolution = K4A_COLOR_RESOLUTION_OFF;`

3. Add `printf ("TimeStamp: %zu\n",k4abt_frame_get_device_timestamp_usec(body_frame))` under `printf("%zu bodies are detected!\n", num_bodies);`

4. Press F5 to compile and run the pragram 

**Expected behavior**
The difference between the current timestamp and the last timestamp should be 33333 (30fps), but the actual result is 66666 (15fps)

**Logs**

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS with Version: 1809.17763.864
 - SDK Version: 
Microsoft.Azure.Kinect.BodyTracking v0.9.4
Microsoft.Azure.Kinect.BodyTracking.Dependencies v0.9.0
Microsoft.Azure.Kinect.BodyTracking.Dependencies.cuDNN v0.9.0
Microsoft.Azure.Kinect.Sensor v1.3.0
 - Firmware version:  
[image]

- CPU:  Intel(R)_Core(TM)_i7-8700_CPU_@_3.20GHz
- Display Card: NVIDIA GeForce 1050TI

**Additional context**

qm13, 2019-12-02:
The NVIDIA GeForce 1050Ti is not sufficient to execute body tracking at 30 fps. If you want to run body tracking at 30 fps we recommend a GeForce 1070.

closed 2019-12-02 by qm13

bastiankayser, 2020-01-15:
Even though this issue is closed, I want to add that I find these requirements way too high. KinectV2 delivered a skeleton tracking easily at 30 fps (with a different method, I know). Still, needing a high end machine to run skeleton tracking at 30 fps defeats the purpose of the Azure Kinect as an device on the edge. Any plans to speed up the method?