microsoft/Azure-Kinect-Sensor-SDK #947: Device not initializing in VMWare Virtual Machine

R0NAM1, 2019-12-02:
(This also applies to the body tracking example Application) In a VMWare virtual machine, sharing the depth camera, 4k camera and microphones, the device fails to start with this log:
`[ warning  ] : is_fw_version_compatable(). Firmware version for Depth Config is 5006.27.0. Consider upgrading to 6109.7.0 or newer.
[ error    ] : depth_engine_start_helper(). Depth engine create and initialize failed with error code: 204.
[ error    ] : deresult == K4A_DEPTH_ENGINE_RESULT_SUCCEEDED returned failure in depth_engine_start_helper()
[ error    ] : depth_engine_start_helper(dewrapper, dewrapper->fps, dewrapper->depth_mode, &depth_engine_max_compute_time_ms, &depth_engine_output_buffer_size) returned failure in depth_engine_thread()
[ warning  ] : capturesync_add_capture(). Capture Error Detected, Depth 
[ error    ] : dewrapper_start(). Depth Engine thread failed to start
[ error    ] : dewrapper_start(depth->dewrapper, config, depth->calibration_memory, depth->calibration_memory_size) returned failure in depth_start()
[ error    ] : cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming()
[ error    ] : depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063
[ error    ] : depth_start(device->depth, config) returned failure in k4a_device_start_cameras()
`
It seems to be an issue with the Depth engine running on virtualized hardware, and that in turn causes the device to not start?

To reproduce:
Install Windows 10 in a VMWare virtual machine, with 3d acceleration enabled, share the depth camera, 4k camera and microphones, open the device in the k4aviewer, press start and see the errors detailed above.

**Expected behavior**
Expected it to work fine in 2d mode, maybe some errors when enabling 3d.
**Logs**
(No log file was produced by the viewer, just in the log box)

**Desktop (please complete the following information):**
 - OS with Version: Windows 10 pro, Ver. 1909
 - SDK Version:  1.3.0
 - Firmware version:  Current Firmware Versions:
  RGB camera firmware:      1.6.102
  Depth camera firmware:    1.6.75
  Depth config file:        5006.27
  Audio firmware:           1.6.14
  Build Config:             Production

rabbitdaxi, 2019-12-02:
To help start the investigation, the error message "Depth engine create and initialize failed with error code: 204." meaning GPU API returns failure based on https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/90f77529f5ad19efd1e8f155a240c329e3bcbbdf/include/k4ainternal/k4aplugin.h#L206

wes-b, 2019-12-02:
A 2nd issue you are hitting is that it looks like you are using pre-production devices:
>[ warning ] : is_fw_version_compatable(). Firmware version for Depth Config is 5006.27.0. Consider upgrading to 6109.7.0 or newer.

I would suggest getting production devices to replace units that use 5006.27.0 config. There will be no firmware update to address this unfortunately, so a new device is needed.

R0NAM1, 2019-12-03:
> 
> 
> To help start the investigation, the error message "Depth engine create and initialize failed with error code: 204." meaning GPU API returns failure based on
> 
> https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/90f77529f5ad19efd1e8f155a240c329e3bcbbdf/include/k4ainternal/k4aplugin.h#L206
> 
> Could be related to how the VM's graphic resources access here?

That was my first feeling, I was using VirtualBox but the program wouldn't even launch because they have basically NO support for OpenGl 1.2 (From what I've read) and so I switched to VMWare.

R0NAM1, 2019-12-03:
> A 2nd issue you are hitting is that it looks like you are using pre-production devices:
>
>
> I would suggest getting production devices to replace units that use 5006.27.0 config. There will be no [...]

So far this has not been a problem for me.

rabbitdaxi, 2019-12-03:
@R0NAM1 I am not familiar with VMWare, I am not sure whether it automatically uses the physical GPU on your host machine. You might need to find some way to check whether in the VM, you can see the GPU hardware. By the way, the depth engine uses directx on windows and opengl on linux, it require a GPU to run for performance reason. You can refer to https://docs.microsoft.com/en-us/azure/Kinect-dk/system-requirements#minimum-host-pc-hardware-requirements for the hardware requirements.

closed 2019-12-12 by wes-b

jesd23, 2019-12-21:
Uploadk #947<https://eur04.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fmicrosoft%2FAzure-Kinect-Sensor-SDK%2Fissues%2F947&data=02%7C01%7C%7Cbf512f5efd8544a2eef908d77f2e192d%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637117708273203943&sdata=RfQSqJPVLwoigsHZkj4q2EB36NKrpenIZkd2qoivCfg%3D&reserved=0>.