microsoft/Azure-Kinect-Sensor-SDK #955: ColorImageToDepthCamera returning Empty Image

jschweda, 2019-12-05:
In general I'm trying to write a .ply file from a capture. XYZ are working quite fine but when I try to add rgb I'm unable to convert the color image to a depth image. 
The Color and Depth images in the Capture are filled but the Transformation.ColorimageToDepthCamera returns the correct image size but all data is 0 0 0.

**To Reproduce**
`

        struct abc
	{
		public byte a;
		public byte r;
		public byte g;
		public byte b;
	}

	class Program
	{
		static void Main(string[] args)
		{
			var device = Device.Open();
			device.StartCameras(new DeviceConfiguration
			{
				ColorFormat = ImageFormat.ColorBGRA32,
				ColorResolution = ColorResolution.R720p,
				DepthMode = DepthMode.NFOV_Unbinned
			});
			var calibration = device.GetCalibration();
			var transformation = calibration.CreateTransformation();
			Capture capture = device.GetCapture();

			while (capture == null || capture.Color == null || capture.Depth == null)
			{
				capture = device.GetCapture();
			}

			int i = 0;
			var colorPixel = capture.Color.GetPixels<abc>();
			Console.WriteLine(colorPixel.Length);
			foreach (var c in colorPixel.Span)
			{
				Console.WriteLine($"{c.a} {c.r} {c.g} {c.b}");
				i++;
				if (i > 200)
				{
					break;
				}
			}

			var depthFromColor = transformation.ColorImageToDepthCamera(capture);
			var depthFromColorPixel = depthFromColor.GetPixels<abc>();
			Console.WriteLine(depthFromColorPixel.Length);
			 i = 0;
			foreach (var c in depthFromColorPixel.Span)
			{
				Console.WriteLine($"{c.a} {c.r} {c.g} {c.b}");
				i++;
				if (i > 200)
				{
					break;
				}
			}

			var depthPixel = capture.Depth.GetPixels<UInt16>();
			Console.WriteLine(depthPixel.Length);
			i = 0;

			foreach (var d in depthPixel.Span)
			{
				if (d > 0)
				{
					Console.WriteLine($"{d} ");
					i++;
				}
				if (i > 200)
				{
					break;
				}
			}

		}
	}
`

**Expected behavior**
From the documentation I would expect argb for all pixels in the transformed image to be filled.

**Desktop (please complete the following information):**
 - OS with Version: Windows 1909 
 - SDK Version: 1.3.0

wes-b, 2019-12-05:
You might want to ask this over in the python repo where you are getting the python wrapper from.

wes-b, 2019-12-05:
Happy to reopen if I missed something and this is an SDK question.

jschweda, 2019-12-06:
This is c# and not python :(

wes-b, 2019-12-06:
Sorry, my apology, I saw .ply and stopped reading.

MsDerekMa, 2019-12-06:
It looks like you are only looking at the first 200 pixels of the transformed depth image. Because of the overlap of the depth and color image, there are several lines of the transformed image that will be 0 as we don't have color data for those points. You can see in this picture the color image on the left and the color to depth transformed image is on the right where all unknown, 0 value, pixels are colored in a light blue. The corners are missing data because the depth sensor in NFOV mode doesn't have depth data for those points, but then the depth sensor does have data above and below the color image causing the gaps at the top and bottom. [image]

Here are some more links if you want more reading about the [depth sensor](https://docs.microsoft.com/en-us/azure/Kinect-dk/depth-camera), [image transformations](https://docs.microsoft.com/en-us/azure/Kinect-dk/use-image-transformation), or [the field of view of the cameras](https://docs.microsoft.com/en-us/azure/Kinect-dk/hardware-specification#camera-field-of-view).

wes-b, 2019-12-07:
Also, a minor observation, you can simply your code by setting SynchronizedImagesOnly = true and you won't have to test for null color or depth images.

closed 2019-12-07 by wes-b