microsoft/Azure-Kinect-Sensor-SDK #960: Use Azure Virtual NVIDIA GPUs to run Body Tracking

benjamminhansen, 2019-12-07:
I was told on [this thread](https://github.com/MicrosoftDocs/azure-docs/issues/44098) that my team can use the Body Tracking SDK on a Windows VM using NVIDIA GPUs. We are working under a university and want to be able to deploy the body tracking application we will create to multiple users to track their physical therapy at home. Thus, they won't all have a compatible GPU. So if we can use a paid Azure resource to share the Windows VM with virtual GPUs to those users, then we can leverage our application to their therapy needs at home. 

My question is, what size of a VM and GPU optimized VM do you recommend? What N-series of NVIDIA VM would you recommend? How much memory does the Body Tracking SDK require, and how intense of graphics is the Body Tracking viewer?

Thank you very much in advance for your help!

qm13, 2019-12-09:
@benjamminhansen this is a very difficult question to answer without understanding the usage model and cost model aspects of your application. Body tracking uses ~60% of a NVIDIA 1070 and 1.9 GB of VRAM. In my analysis VRAM tends to be the limiting factor. Theoretically you should be able to host 16 trackers on a V100 VM but I must caveat that we have not tested this. Your mileage may vary depending on your application.

benjamminhansen, 2019-12-10:
Thank you @qm13, that was very helpful for us. We will use this information. I will open a new issue as the need arises.

closed 2019-12-10 by benjamminhansen