microsoft/Azure-Kinect-Sensor-SDK #963: Undistort example fails to create look up table with v1.3.0

chrisfromwork, 2019-12-10:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/17b644560ce7b4ee7dd921dfff0ae811aa54ede6/examples/undistort/main.cpp#L394

The above line in the undistort example fails to create the lookup table k4a image on my pc. Calls to create other image types (for example, the undistorted image) succeed. The pinhole values seem to be valid, both the width and height are 512. I am using a download of the v1.3.0 Azure Kinect SDK.

wes-b, 2019-12-10:
Can you share logs generated by the SDK for the failure?

chrisfromwork, 2019-12-14:
yes, are there instructions for how to obtain these logs?

chrisfromwork, 2019-12-14:
i was able to get source to resolve with pdbs, It turned out i was building against one version of the sdk but was calling into a different version at runtime. This caused image types to get out of sync causing image creation failures.

closed 2019-12-14 by chrisfromwork