microsoft/Azure-Kinect-Sensor-SDK #964: Body Tracking is inaccurate in WFOV depth mode

lydiakimhehe, 2019-12-10:
**Describe the bug**

Hello, 
I'm trying to use body tracking in many scenario. 
Mostly, in <K4A_DEPTH_MODE_NFOV_UNBINNED> mode, body tracking works well. 
But the resolution is quite narrow so that it is hard to get the wider depth image that I expected.
(because body tracking does not work properly out of depth image)  

Therefore,  there are some parts that I can see in the colour image, but not in depth image. 

Because of this reason, I'm trying to use body tracking in <K4A_DEPTH_MODE_WFOV_UNBINNED> mode which gives the wider resolution of depth image. 

Then in this <K4A_DEPTH_MODE_WFOV_UNBINNED> depth mode, body tracking does not work properly. 

Is there any reason for this or just me experiencing this? 

Thank you 

**To Reproduce**

**Expected behavior**

I'd expect to have body tracking work accurately in <K4A_DEPTH_MODE_WFOV_UNBINNED> mode. 

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version:  Ubuntu 18.04
 - SDK Version:   1.3.0
 - Body Tracking SDK Version: 0.9.5 
 - Firmware version:    1.6

**Additional context**

qm13, 2019-12-11:
@lydiakimhehe You should be seeing similar results in WFOV to NFOV. Could you please post imagery showing the issues you are seeing?

qm13, 2019-12-17:
Also note that WFOV has a smaller operating range than NFOV. See https://docs.microsoft.com/en-us/azure/Kinect-dk/hardware-specification.

qm13, 2020-01-06:
No repro

closed 2020-01-06 by qm13