microsoft/Azure-Kinect-Sensor-SDK #965: Color.SystemTimestampNsec looks bad over time.

markobe-msft, 2019-12-10:
**Describe the bug**

I have been looking at AK timestamps in order to understand how to properly setup sync with another device. I've noticed when using device time (Depth.DeviceTimestamp.Ticks and Color.DeviceTimestamp.Ticks) depth and color camera periods and offset are consistent. On the other side, when using **system time (Depth.SystemTimestampNsec and Color.SystemTimestampNsec). Depth camera period is consistent, while Color period is jumpy and offset is diverging until it resets.**

Please check following charts, I hope you find them helpful. 
[image]
[image]

I assume there is something going wrong with how device timestamp of the color camera is being converted to the system time, as color camera period calculated with system time jumps to ~48ms every ~12 frames. 

As temporary workaround, I will rely on the depth camera system timestamp. 

**To Reproduce**

1. Visual Studio -> C# new console project
2. Right click project, go to manage nuget packages, Click Browse, search for Microsoft.Azure.Kinect.Sensor
3. Install SDK 1.3.0
4. Paste following code, compile and start
      ```static void Main(string[] args)
        {
            using (var dev = Device.Open())
            {
                dev.StartCameras(new DeviceConfiguration()
                {
                    CameraFPS = FPS.FPS30,
                    ColorFormat = ImageFormat.ColorBGRA32,
                    ColorResolution = ColorResolution.R720p,
                    DepthMode = DepthMode.WFOV_2x2Binned,
                });
                dev.SetColorControl(ColorControlCommand.ExposureTimeAbsolute, ColorControlMode.Manual, 2000);

                long previousDepthDeviceTimestamp = 0;
                long previousColorDeviceTimestamp = 0;
                long previousDepthSystemTimestampNsec = 0;
                long previousColorSystemTimestampNsec = 0;

                while (true)
                using (var cap = dev.GetCapture())
                {
                        Console.WriteLine("DepthTickUs, {0}, ColorTickUs, {1}, DepthToColorTickUs, {2}, DepthTimeUs, {3}, ColorTimeUs, {4}, DepthToColorTimeUs, {5}, DepthExposureMs, {6}, ColorExposureMs, {7}, DepthDiffTickUs, {8}, ColorDiffTickUs, {9}, DepthDiffTimeUs, {10}, ColorDiffTimeUs, {11}"
                            , (cap.Depth != null) ? cap.Depth.DeviceTimestamp.Ticks / 10 : 0
                            , (cap.Color != null) ? cap.Color.DeviceTimestamp.Ticks / 10 : 0
                            , (cap.Depth != null) && (cap.Color != null) ? (cap.Depth.DeviceTimestamp.Ticks - cap.Color.DeviceTimestamp.Ticks) / 1000 : 0
                            , (cap.Depth != null) ? cap.Depth.SystemTimestampNsec / 1000 : 0
                            , (cap.Color != null) ? cap.Color.SystemTimestampNsec / 1000 : 0
                            , (cap.Depth != null) && (cap.Color != null) ? (cap.Depth.SystemTimestampNsec - cap.Color.SystemTimestampNsec) / 1000 : 0
                            , (cap.Depth != null) ? cap.Depth.Exposure.Milliseconds : 0
                            , (cap.Color != null) ? cap.Color.Exposure.Milliseconds : 0
                            , (cap.Depth != null) ? (cap.Depth.DeviceTimestamp.Ticks - previousDepthDeviceTimestamp) / 10 : 0
                            , (cap.Color != null) ? (cap.Color.DeviceTimestamp.Ticks - previousColorDeviceTimestamp) / 10 : 0
                            , (cap.Depth != null) ? (cap.Depth.SystemTimestampNsec - previousDepthSystemTimestampNsec) / 1000 : 0
                            , (cap.Color != null) ? (cap.Color.SystemTimestampNsec - previousColorSystemTimestampNsec) / 1000 : 0
                            );

                        if (cap.Depth != null)
                        {
                            previousDepthDeviceTimestamp = cap.Depth.DeviceTimestamp.Ticks;
                            previousDepthSystemTimestampNsec = cap.Depth.SystemTimestampNsec;
                        }
                        if (cap.Color != null)
                        {
                            previousColorDeviceTimestamp = cap.Color.DeviceTimestamp.Ticks;
                            previousColorSystemTimestampNsec = cap.Color.SystemTimestampNsec;
                        }
                    }
            }
        }
    }

**Expected behavior**

I would expect depth and color camera periods and offset are to be consistent when using sytem time (Depth.SystemTimestampNsec and Color.SystemTimestampNsec). 

**Desktop (please complete the following information):**
 - OS with Version: Windows 1909
 - SDK Version:  1.3.0
 - Firmware version: 
Device Serial Number: 000264285112
Current Firmware Versions:
  RGB camera firmware:      1.6.98
  Depth camera firmware:    1.6.70
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft

wes-b, 2019-12-10:
We expect device timestamp to be the most stable and periodic because it is stamped by hardware at the center of exposure (single device mode). System timestamp in the current SDK will be jumping and highly dependent on USB bandwidth as the system timestamp is recorded with the last arriving USB packet. In the future we hope to make the system timestamp more accurate.

Documentation for [system timestamp]( https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_ga98ba229d1ee1cf3b7c27a0f95e14826b.html#ga98ba229d1ee1cf3b7c27a0f95e14826b) and [device timestamp]( https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_ga318e115976ac441b513f6cf3c102a655.html#ga318e115976ac441b513f6cf3c102a655). Please let us know if there is anything that is unclear in the documentation. 

There is an [issue on Windows]( https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/966) where the system timestamp is not being applied as expected.

I also see the jump from 32ms to 48ms, which implies the system timestamp on color is using MSFT’s 15.6ms clock tick rather than the high precision clock. The jump would just be the rounding error based on the low precision of the time stamp.

markobe-msft, 2019-12-11:
Thanks for swift reply! 

> We expect device timestamp to be the most stable and periodic because it is stamped by hardware at the center of exposure (single device mode).
Could you please tell what's the expected behavior of the synchronization in the external trigger mode?

> "... as the system timestamp is recorded with the last arriving USB packet. "

I'm familiar with this approach. Could you please disclose what's the estimated transfer time for depth image. Byte size of the depth image, and usb packet size would be super helpful. I would love to know when the actual exposure happened in the host side time.

wes-b, 2019-12-12:
>Could you please tell what's the expected behavior of the synchronization in the external trigger mode?

When using multiple devices, the device time stamp captured by firmware is at the start of exposure instead of the center of it. There are a couple bugs in firmware, so you need to most recent FW in the 1.4 alpha Nuget package. And there are more fixes still pending.

> Could you please disclose what's the estimated transfer time for depth image. Byte size of the depth image, and usb packet size would be super helpful. I would love to know when the actual exposure happened in the host side time.

Most of this is in code, we just need to go find it. Transfer time is harder because for depth it is dependent on other USB traffic. See [this](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/95f1d95f1f335b57a350a80a3a62e98e1ee4258d/src/depth_mcu/depthcommands.h#L26) for depth image sizes.

closed 2019-12-18 by wes-b