microsoft/Azure-Kinect-Sensor-SDK #968: How to get/output depth information of each pixel using Azure Kinect?

HaipengU, 2019-12-11:
I am very new to Kinect and I am working on measuring the volume of the object using Kinect. To accomplish this, I need to get the depth value of each pixel which comprised the object. I read some tutorials about Kinect v2 and it seems the depth values are saved as a singular array with the index as x + y * width, where x and y are the pixels coordinates in 2D dimension. 

Any suggestions are appreciated. Thx.

wes-b, 2019-12-12:
The depth image is an array of uint16_t's. It represents the depth in mm's. The [fast point cloud](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/a227f73b8c78e17db843e7a962a1fabd571c6213/examples/fastpointcloud/main.cpp#L54) example works with depth in the Azure Kinect domain, while the [green screen](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/a227f73b8c78e17db843e7a962a1fabd571c6213/examples/fastpointcloud/main.cpp#L54) example uses it in the OpenCV domain.

rabbitdaxi, 2019-12-16:
@HaipengU Thank you for opening the git hub issue, yes, the depth image is contains width * height of uint16_t pixel values, e.g. you can see the documentation of the image format description [here](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___enumerations_gabd9688eb20d5cb878fd22d36de882ddb.html#gabd9688eb20d5cb878fd22d36de882ddb) (K4A_IMAGE_FORMAT_DEPTH16). The buffer store the image as row-major order, you can read the depth from the buffer by first cast the buffer to uint16_t*, then for a pixel with indexing '[i, j]', you can use 'j * width + i' to get the depth value from the uint16_t* buffer

closed 2019-12-16 by wes-b

HaipengU, 2019-12-17:
> @HaipengU Thank you for opening the git hub issue, yes, the depth image contains width * height of uint16_t pixel values, e.g. you can see the documentation of the image format description [here](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___enumerations_gabd9688eb20d5cb878fd22d36de882ddb.html#gabd9688eb20d5cb878fd22d36de882ddb) (K4A_IMAGE_FORMAT_DEPTH16). The buffer store the image as row-major order, you can read the depth from the buffer by first casting the buffer to uint16_t*, then for a pixel with index `[i, j]`, you can use `j * width + i` to get the depth value from the uint16_t* depth buffer

Dear Andrew, thank you for the reply and your suggestions are very helpful!

HaipengU, 2019-12-17:
> The depth image is an array of uint16_t's. It represents the depth in mm's. The [fast point [...]

Thx!