microsoft/Azure-Kinect-Sensor-SDK #969: Capture out of synce when set subordinate_delay_off_master_usec

Krisonrik, 2019-12-11:
**Describe the bug**
We are recording with 8 Kinect with a single computer. When normally recording with 0 delay, everything is synchronized (checked by using flash light). But I realized the interference issue when recording this way so I added a delay for each subordinate Kinect with value that is multiple of 0.16 (0.16ms = 160us). But now cameras go out of sync. (checked with flash light, timing across different color capture can differ as much as 3-4 frames).

**To Reproduce**
1. Assign non-zero value to subordinate_delay_off_master_usec of k4a_device_configuration_t that's being passed into k4a_device_start_cameras. 
2. Observe frame captured go out of sync.

**Expected behavior**
When added a delay, all Kinect frames should still be synced.

**Desktop (please complete the following information):**
 - OS with Version: Windows 10 version 10.0.17763
 - SDK Version:  1.3.0
 - Firmware version:  1.6.102075014

Krisonrik, 2019-12-12:
I realized value for delay should be multiple of 160. I think I was thrown off by some of the other threads talking about offset in milliseconds. Still after corrected this error, in a one minute capture, there are random cameras that would be out of sync by 1-2 frames.

wes-b, 2019-12-12:
Have you looked at issue #530? You should upgrade to the most recent firmware, but you will also want to get the next version that is currently private and being tested. See #951 for the Nuget to access the firmware file in alpha release.

Krisonrik, 2019-12-12:
I have looked into #530, this is off sync more than 100us by a lot. Also, since the #951 is regarding a buggy alpha release, I am concerned to try to load a buggy firmware onto device. From the Nuget list, 1.4 alpha is only one release ahead of 1.3 which is what I have right now. I'm as up to date as I can in stable release point of view.

Krisonrik, 2019-12-12:
Also, this issue was first discovered when I try to use 0 subordinate_delay_off_master_usec  and have delay values on depth_delay_off_color_usec. That cause off sync as well. It's very puzzling, If every camera should be triggered at the same time, why would depth capture delay throw the entire capture off sync?

Krisonrik, 2019-12-12:
In regards to 1.4, I saw it could cause unresponsive kinect. If I upgrade the firmware, can I down grade to 1.3 if it's causing other issues for me?

wes-b, 2019-12-12:
The alpha firmware 1.6.104075014 only causes the documented issue in rare cases. It is also easy to down grade the firmware.

wes-b, 2019-12-12:
1.6.104075014 does contain quite a few multi device synchronization fixes. There are still a few open issues that we hope get fixed soon.

Krisonrik, 2019-12-13:
after upgrading to 1.6.104075014, it was still randomly out of sync. But after some further multi-threading optimization, it seems to be syncing for now.  I'm still very puzzled why adding delay would introduce this issue. Also, I disabled auto exposure, because I read in a few tickets that repeated exposure adjustments can lead to out of sync. May I ask how big is the buffer inside Kinect?  I'm wondering whether this behavior is due to not calling get_capture fast enough. But I also didn't see any error message in regards to buffer overflow or memory leak.

wes-b, 2019-12-17:
For color and depth we will hold upto 8 captures before we drop one due to the not reading fast enough.

And correct for multi device synchronization you should use manual exposure.

wes-b, 2020-03-13:
This is fixed in 1.6.108079014 and availiable in our beta, soon to be official release: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1093.

closed 2020-03-13 by wes-b