microsoft/Azure-Kinect-Sensor-SDK #971: Multiple Kinect devices connected to min requirements PC do not work properly

tesych, 2019-12-12:
**Describe the bug**
When trying to connect multiple Azure Kinect DK devices to the [single min requirements PC](https://docs.microsoft.com/en-us/azure/Kinect-dk/system-requirements#minimum-host-pc-hardware-requirements), see significant frames drops and high impact on the GPU.
What PC to use with multiple devices?

**Desktop (please complete the following information):**
 Windows 10
 7th Gen Intel® CoreTM i3 Processor (Dual Core 2.4 GHz with HD620 GPU or faster)
 4GB Memory
Dedicated USB3 port

tesych, 2019-12-13:
During the Microsoft ignite conference we created a volumetric capture experience booth in partnership with [Scatter](http://depthkit.tv) and their DepthKit. 

**The set up included:**
- 5 Azure Kinect DK devices (tested with 10 with no issues)
- For USB 3.0 Data extension, the Cable Matters 5 Meter Active USB 3.0 extension
- For power, the Monoprice 5 Meter USB 2.0 extension and the power adapters that come with the Azure Kinect
**-PC**
     Intel i9 9920x Extreme (3.5GHz,12-core / 24-thread)
     32GB DDR4 memory (8GB x 4) 
     NVIDIA RTX2080Ti video card
     Asus X299Tuf MK2 Motherboard
     3x StarTech.com 4 Port USB3.0 PCIe Cards with 4 Dedicated 5Gbps Channels
**- Data steaming options**
     720p color
     640x576 depth
**- Software** running on the PC**
    - Sensor SDK
    - Demo app that was built in the Unity game engine
    - Plugin for Unity that supports the Unity VFX graph for particle system effects
    - [DepthKit](http://depthkit.tv)

**CPU/ GPU usage** 
With 10 devices at these settings, the CPU utilization was around 43% and the GPU utilization was around 50% with zero dropped frames; with only 5 Devices, utilization is roughly half of that. In this scenario the work being done by Depthkit is an optimized pipeline that handles frame acquisition from the device, frame synchronization, 3D viewport rendering of the depth and color data, as well as local GPU texture memory sharing to other applications, like the Unity based Ignite demo app.

UnaNancyOwen, 2019-12-13:
@tesych Thank you for share it. I have one question. (Sorry, that question is not related to main problem of this issue.)

> For power, the Monoprice 5 Meter USB 2.0 extension and the power adapters that come with the Azure Kinect

Is it active extension type cable (https://www.monoprice.com/product?p_id=6149) or non-active extension type cable (https://www.monoprice.com/product?p_id=5432)?

tim-depthkit, 2019-12-13:
Hi @UnaNancyOwen 

I can speak to this as a Scatter teammate involved in this production. We used an active USB 2.0 extender for power, [https://www.amazon.com/Monoprice-Extension-Repeater-PlayStation-Keyboard/dp/B009GUTFX8/](this one specifically)

UnaNancyOwen, 2019-12-14:
@tim-depthkit Thanks! 👍

phg1024, 2020-01-14:
@tesych Wanted to confirm with you, the setup you mentioned is a single PC to drive 5 Kinects, right?

tesych, 2020-01-17:
@phg1024 it is correct. One host with 5 Azure Kinect devices.

closed 2020-01-19 by tesych

SaraHackett13, 2020-09-29:
Is [this cable](https://www.amazon.com/Cable-Matters-Active-Extension-Female/dp/B00DMFB5OK/ref=sr_1_3?dchild=1&keywords=Cable+Matters+5+Meter+Active+USB+3.0+extension&qid=1601412361&s=electronics&sr=1-3) the same USB 3.0 data extension used in your testing? 

If so, did you power each extension cable as well as powering each Kinect camera?

tim-depthkit, 2020-09-30:
@SaraHackett13 Yes that is the one! We did not need to power the data cable, but we did use a separate USB extender to extend the Azure Kinect's factory power cable.

MaxPalmer-UH, 2021-02-23:
In this setup was data being recorded? If so, how did the system cope with data being written to disk as the number of sensors increased? Did it become a limiting factor?

Ahmed-KASAPBASI, 2021-09-30:
I have the same question but for kinect v2, I want to connect 2 or 3 Kinect v2 to one Computer?